MarlinFirmware/Marlin #20356: [FR] (feature request title)

elmek6, 2020-12-02:
### Description

I don't like to print ABS nor PETG. It gives me always hard time. Today I print same part 4 times as you can see 2 big clamp hold it for potential warping but the biggest problem after printing %80 of object, there is a small bridge, I know at this point my print will collapse or make spaghetti.
I modified my printer and put additional 1 pot and 1 switch. This switch selecting the manual or auto fan mode. And pot adjust the fan speed %0 - %100. Sometimes you need to override realtime fan speed. Slicer won't offer many choice or you make mistake but you won't discover until too late.

### Feature Workflow

I wish there was a settings in Marlin Menu like "Override fan speed" and GCode for it, like "M106 S0 O"
M106 [I<index>] [P<index>] [S<speed>] [T<secondary>] [O<flag>]

Example:
M106 S0 O ;turn off fan, until job done or job restart  or reset

#### Additional Information

See attachment.

hamster65, 2020-12-02:
This is something I wish for as well. It often happens that I start a print with ABS and then I realize that I have forgotten to disable the fan. I pause the print and disconnect the fan, but then the contacts are unprotected and prone to short circuits touching other metal components.

Simply setting the fan to zero from the tuning menu does not work because this setting is not working as an offset, so new gcode commands turn the fan back on.
Speaking of offsets: It would be a nice feature to have offsets in the tuning menu for fan and temperatures.  Temperature offsets are already in Octoprint, but a fan override/offset is not and my FR there was not successful.

elmek6, 2021-01-01:
I made something
https://github.com/elmek6/Marlin/blob/2.0.x/Marlin/src/gcode/temp/M106_M107.cpp