MarlinFirmware/Marlin #20358: [BUG] Wrong answer to M851 - Probe offset

radek8, 2020-12-02:
I have a BLTouch probe, Probe offset is set in configuration.h. The display shows the value correctly, but when asked via the serial port, it answers axis X 0.0000 and Y 0.0000 . The Z axis is OK

### Steps to Reproduce

1. sent via serial port M503 or M851

**Expected behavior:**

The correct values are displayed

**Actual behavior:**

SENDING:M851
Probe Offset X0.0000 Y0.0000 Z-1.1100

M503 S0
M851 X0.0000 Y0.0000 Z-1.1100 

#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5631854/config.zip)

thisiskeithb, 2020-12-02:
You're missing some config options from the latest `bugfix-2.0.x`, please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

radek8, 2020-12-02:
I downloaded the latest edition
last PR20241.
The problem remains

sjasonsmith, 2020-12-02:
Sounds like EEPROM needs to be initialized. It probably has zeroes stored.

thisiskeithb, 2020-12-02:
> last PR20241

That's not the latest. The latest was a4d6908d556afd77cf66ef7a8adc7c34fee81674.

There was a PR put in after to right after to fix the probe X/Y offset that followed right after that. Again, 
please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

radek8, 2020-12-02:
Sorry the compilation with PR20241 was the previous one.
Current build a4d6908d55
The problem is fixed.
Thank you.

closed 2020-12-02 by radek8