MarlinFirmware/Marlin #20359: [BUG] LPC176x (SKR 1.3) boards and MAX6675 not working

GadgetAngel, 2020-12-02:
### Bug Description

Also see issue https://github.com/MarlinFirmware/Marlin/issues/19417#issue-703065382.   It has been closed.

I set up my SKR V1.3 board to use a MAX6675 board with an E3D K-Type Thermocouple.  I checked the thermocouple and I am certain the thermocouple is isolated from the cartridge tip and from the T+ and T- pins.  I place a 0.80 uF capacitor across the T+ and T- leads to reduce noise in the readings. Originally it was only 0.33 uF.

My SKR V1.3 has one extruder. It uses TMC2130 in SPI mode on X-axis, TMC2226 in UART mode on Y-axis, TMC2226 in UART mode on Z-axis and TMC2226 in UART mode on Extruder. I used a BTT TFT35 V3.0 screen in TFT mode only so I could free up the EXP1 and EXP2 pins on the board for testing Hardware SPI and Software SPI.

In pins_BTT_SKR_common.h file:
```
    #define TEMP_0_PIN                        P2_06
    #define MAX6675_DO_PIN                    P2_02         
    #define MAX6675_SCK_PIN                   P1_17        
    #define MAX6675_SS_PIN                    TEMP_0_PIN  
```
In configuration.h file:
```
#define EXTRUDERS 1
#define TEMP_SENSOR_0 -2
//#define THERMAL_PROTECTION_HOTENDS
//#define THERMAL_PROTECTION_BED
//#define THERMAL_PROTECTION_CHAMBER
```
on first attempt In configuration_adv.h file:
```
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED            10 
#define MILLISECONDS_PREHEAT_TIME                                30000    //30 seconds delay
```
on last attempt In configuration_adv.h file:
```
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED            200 
#define MILLISECONDS_PREHEAT_TIME                                60000    //1 minute delay
```

When I connect a MAX6675 board up to LPC176x motherboard I get an immediate MIN TEMP Error! If I set MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED and MILLISECONDS_PREHEAT_TIME in configuration_adv.h file to **really** unreasonably high numbers, I still get an immediate MIN TEMP Error.   So I go into temperature.cpp and enable `#define IGNORE_THERMOCOUPLE_ERRORS` to see what is going on.  

I have more than one MAX6675 board.  

Here are some samples of what is going on with the MAX6675 board in Software or Hardware SPI mode (the board sends 0).
[image]

The board will read the temperature but then in a couple of samples it will send an hard 0.  

After investigating this, the MAX6675 takes 250 milliseconds at the maximum to do its conversion.  Temperature::read_max6675 uses the Marlin variable MAX6675_HEAT_INTERVAL (which really is the SAMPLE_INTERVAL) to only request the temperature reading from the SPI MAX6675 every 250 milliseconds.  Some sources say 300 milliseconds would work better so I tried to set MAX6675_HEAT_INTERVAL to 300 and even set up a way to measure the timing between temperature request. Marlin was doing a request every 309 or less.  Close enough, because my additional statements would throw off the measurement.  

So I decided it was not the timing between reads that was getting me. 

I check the code and it looks good, except the SKR V1.3 board does not let you use E1 DIR_PIN (P0_00) and E1 STEP_PIN (P0_01) for TEMP_0_SENSOR.  Once I corrected the code in Marlin\src\HAL\LPC1768\inc\SanityCheck.h file to allow P0_00 and P0_01 for the SKR V1.3 board, I still got an immediate MIN TEMP error.  So I decided to rename the X driver Socket PINS in pins_BTT_SKR_common.h to be E1 driver Socket PINs and renamed the previous E1 driver Socket PINS to X driver Socket PINS to see if the problem followed along with the driver socket change. I ran into a similar issue with the MAX31865 board on the SKR V1.3 board. 
```
Side note:  When I used the E1 DIR_PIN and E1 STEP_PIN and E1 CS_PIN with the MAX31865 PT100 board on SKR V1.3, I get a
 read of hard 0 from Adafruit_MAX31865::readRTD() call.  So I thought maybe I was running into that problem.  After 
changing the code around in temperature.cpp I was able to get the MAX31865 PT100 board to work on the SKR V1.3 board.  I 
am still testing out my fix but if it works, I will be submitting a PULL REQUEST to fix this error and the P0_00 and P0_01 
SanityCheck.h error.
```
Back to this issue: I now have X driver Socket open with the E1 driver socket acting as my X-Axis.  This way I will not get the SKR V1.3 SanityCheck.h error about using P0_00 or P0_01 because the DIR_PIN is now P2_06 and the STEP_PIN  is now P2_02.  Marlin compiles but I still get 0 every once in a while with the MAX6675.  I know I do not have a ground loop.  So I hooked up another MAX6675 board.  Same issue.  So I order three new thermocouples from E3D but they have not yet arrived.  I turned off all checks in Marlin for Thermal Runaway protection and add code to print values to the SERIAL_PORT via Pronterface.  After using my heat gun and watching the temperature value printing to the SERIAL_PORT I concluded that the MAX6675 board is a poor amplifier board.  I think the firmware that reads the internal ADC is flawed.  

So I decide to code up a work-around to the problem.  I added a small amount of code to do sensor validation of the temperature data being read.  It looks at the adjacent samples and with a specified SPREAD between points throws away the bad data.  Temperature readings should not go from a value of 800 and then in the next sample go to 0.  It works and I will submit it as a PULL REQUEST for your consideration.  I want to check the code with the SKR PRO board and the GTR V1.0 board before I submit the PR.

If the sensor validation code was going to drop a temperature sample, I printed the two temperature samples and the calculated SPREAD or DIFF to the SERIAL_PORT. the MAX6675_SPREAD_BETWEEN_ADJ_SAMPLES is set for 40 in this example. I used a heat gun to raise the temperature on the K-type thermocouple. Here is a small sample of the output:

**max6675_temp_prev[hindex]** is the previous temperature sample taken, **max6675_temp2** is the current temperature sample. **max6675_start_cntr for MAX6675_START_SAMPLESZ = (11)** tells me the beginning buffer size was reached  because the MAX6675 CHIP needs to stabilize a bit.  The buffer size is 10 samples which is set by MAX6675_START_SAMPLESZ.  The Boolean flag **max6675_init** being 0 indicates the sensor validation initialization condition was meet and the validation routine has started.
```
Connecting...
Printer is now online.

max6675_temp_prev[hindex] = (115)  max6675_temp2 = (0)  DIFF = (115)
max6675_start_cntr for MAX6675_START_SAMPLESZ = (11)  boolean flag max6675_init = (0)
================================================

[... 193 lines cut ...]
Disconnected.

```
As you can see from the data the K-Type Thermocouple is NOT linear and that most of the outlying data is 0.  When the temperature was getting hot, you do get dips in the data.  So, basically my sensor validation code smooths out the curve so you do not see these **unreasonable** temperature samples.  I get these 0 values if I run MAX6675 in Software or Hardware SPI mode.   

### Configuration Files

[Configuration_MAX6675.zip](https://github.com/MarlinFirmware/Marlin/files/5632548/Configuration_MAX6675.zip)

### Steps to Reproduce

1. Hook up a MAX6675 board to a K-Type Thermocouple from E3D online
2. set TEMP_SENSOR_0 to -2 in configuration.h file
3. in the pins file correctly setup the MAX6675_SS_PIN to point to the CS pin on the MAX6675 board if using Hardware SPI and ensure you hook up the DO pin and SCK pin on the MAX6675 board to the EXP2's hardware default SPI bus. If using software SPI define 
MAX6675_DO_PIN, and MAX6675_SCK_PIN in the pins file to point to the pins on EXP1 or to an empty driver socket pins.

In my case, I used DIR_PIN and STEP_PIN because I had TMC2130 in SPI mode and TMC2226 in UART mode. If you use the empty driver socket just ensure you place a UART jumper on the empty drive socket's UART Jumper pad for SKR V1.3. With the UART jumper in place, you can use the empty driver's socket CS_PIN for the SS_PIN for the MAX6675 board.  I could not use the empty driver socket MS1, MS2, or RST PINS because they were being used by the TMC2130 in SPI mode.

**Expected behavior:**

I expected the K-type Thermocouple to read the temperature.  I also, expected the configuration_adv.h Marlin variables MILLISECONDS_PREHEAT_TIME set at 200 and MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED set for 1 minute to cause a delay in the MIN TEMP error but it did not.  I know those are really large values. I wanted to see if I was getting any data from the MAX6675 board (I knew the MAX6675 was producing data) but the data was going to 0 also.  So why did I not see a delay in the Marlin code before producing the MIN TEMP error?

**Actual behavior:**

See my above discussion.

rhapsodyv, 2020-12-03:
I purchased a MAX31865, to tackle those max spi issues soon. If MAX6675 is too different, I can order one too. But I ordered it from Ali, so it will take about a month to arrive.

GadgetAngel, 2020-12-03:
@rhapsodyv the MAX6675 is totally different than either the MAX31865 or MAX31855. The MAX6617 as I have shown is sending “0” samples or the timing is just slightly off due to the DELAY_NS bug and the MCU is returning with a 0 because nothing came across. I do not own a logic analyzer yet so I can not tell if the proper SPI lines are being set.  If I were you I would order a MAX6675 board also

GadgetAngel, 2020-12-03:
@rhapsodyv Do you want me to submit the work-around to Marlin as a PR?

rhapsodyv, 2020-12-03:
>  If I were you I would order a MAX6675 board also

Ok, just ordered a MAX6675 too.

> @rhapsodyv Do you want me to submit the work-around to Marlin as a PR?

Sure, let's see what we can do.

I was working on it, with another user about a month ago, but the user burned his sensor... then he stopped replying and everything cool down.

rhapsodyv, 2020-12-03:
Ahh, and you is the author of #20132. That PR is the reason that I brought the MAX31865

Grogyan, 2020-12-07:
The MAX6675 is considered obsolete, the MAX31855 replaces it, which I have been using for years without issue

GadgetAngel, 2020-12-08:
@rhapsodyv tomorrow I will open another bug report, the MAX31855 works in Hardware SPI mode on LPC176x boards but not in software SPI mode.  The LPC176x boards reads zero every once in a while which causes the printer to hit MIN TEMP error.  So you better purchase yourself a MAX31855 board also.  I have some capture data files that you will be able to look at from my Saleae 8 channel logic Analyzer.  You can download their free software from saleae.com/downloads.  This way you can see what is going on with the SPI signaling.  I do have a work-a-round I will submit in PR once I get the bug report done.  

@Grogyan I can except the fact that the MAX6675 is obsolete, so why doesn't Marlin drop it as an option with the software?  If it is an option, I will test it to see if it is not functioning and try to get it to work.  

@rhapsodyv After I submit the PR to fix the MAX31855 on LPC176x boards, I am going to go back to look at the MAX31865.  Maybe once you see the problems with Software SPI and the MAX31855 we can figure out the issue and it might straighten out the issues with the other SPI boards and the LPC176x MCU boards.

GadgetAngel, 2020-12-11:
@rhapsodyv After further testing of the MAX6675 board, I found an issue with the SKR V1.3 using hardware SPI and utilizing the AUX-1 connector signal pins as CS.  The Current Marlin-bugfix branch does not all the SKR V1.3 to use TX0 or RX0 as CS in hardware SPI mode.  

In other words, If In pins_BTT_SKR_common.h file:
```
    #define TEMP_0_PIN                        P0_02 or P0_03
    #define MAX6675_SS_PIN                    TEMP_0_PIN
```
ensure that in pins_BTT_SKR_V_3.h file:
```
//
// SD Support
//

#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION                  LCD
#endif
```
In configuration.h:
```
//define SERIAL_PORT_2 0 
#define EXTRUDERS 1
#define TEMP_SENSOR_0 -2
//#define THERMAL_PROTECTION_HOTENDS
//#define THERMAL_PROTECTION_BED
//#define THERMAL_PROTECTION_CHAMBER
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
In configuration_adv.h:
```
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED            10
#define TMC_USE_SW_SPI 
#define MONITOR_DRIVER_STATUS 
#define TMC_DEBUG
#define THERMOCOUPLE_MAX_ERRORS 15
```
Then a physically tap into the EXP2 connector's and do the following hardwire connections : 
```
P0_17 and wire that to MAX6675's DO pin 
P0_15 and wire that into the MAX6675 's SCK pin
and connect either P0_02 or P0_03 on the AUX-1 or the TFT connector to the MAX6675's CS pin
```

**I get MAX TEMP error! NOT a MIN TEMP but a MAX TEMP error.** 

So I enable `#define IGNORE_THERMOCOUPLE_ERRORS` in temperature.cpp code and find out that the reason for the MAX TEMP error is because the MAX6675 is reading and OPEN THERMOCOUPLE and set the temperature reading to the maximum to trigger a MAX TEMP error.  

So I switch the MAX6675_SS_PIN to P1_01 pin and recompile and it reads the temperature without an MAX TEMP error.  So now I know that the hook up to the hardware SPI is done right.  

So I go back and hook up MAX6675_SS_PIN to P0_02, again I immediately get MAX TEMP error "Open Thermocouple" detected. So I switch to P0_03 and get the same result. 

The thermocouple connection is just fine.  I look at the SPI capture of the communication and the SPI signaling is messed up when using P0_02 or P0_03 in hardware SPI mode.  The "Open Thermocouple" is because the LPC framework is reading the data wrong!  

I developed a software library call and tried it again after working out the bugs in my code.  Guess what?  With the library call for hardware SPI the P0_02 and P0_03 pins can be used with hardware SPI on the SKR V1.3 board.

I want to address the issue in which I opened this BUG report.  This BUG report should now be the BUG I am reporting to you now. I will call it "SKR V1.3 board and MAX6675 not working in hardware SPI with AUX-1 P0_02 and P0_03 PINS".  

After working on this BUG (SKR V1.3 board and MAX6675 not working in hardware SPI with AUX-1 P0_02 and P0_03 PINS) and after realizing that not having the heater_on was not allowing Marlin to use the `MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED` directive properly.  

I modified the code to allow Marlin to use `MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED` even if the heater was off.  

After doing this I never again saw the zero temperature drops with the MAX6675 because Marlin was now allowed to handle the situation as it should.  So because the issue on this BUG report no longer applies, I will change the title of this one so I do not have to fill out another BUG report.

GadgetAngel, 2021-01-12:
> This issue has had no activity in the last 30 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 7 days.

This issue is still under review because the pull request that fixes this issue is still going through the review process.

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by GadgetAngel