MarlinFirmware/Marlin #20362: [BUG] Cannot compile for SKR mini E3 on fresh VScode/Git install (Windows)

Taomyn, 2020-12-03:
### Bug Description

I needed to reinstall my VSCode/Git environment on my laptop today and noticed that my normal compile of Marlin for my SKR mini E3 v1.2 kept failing because it could not find `SoftwareSerial.h` to include. After some digging around and also comparing the environment to my other PC which would build just fine, it was because it was not downloading the lib dependency "SoftwareSerialM" which the other machine still had an old copy of.

So after adding `lib_deps          = ${common_stm32f1.lib_deps}` to the `[env:STM32F103RC_btt]` section of platformio.ini it worked again after it downloaded the files.

```
[env:STM32F103RC_btt]
platform          = ${common_stm32f1.platform}
extends           = env:STM32F103RC
extra_scripts     = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
build_flags       = ${common_stm32f1.build_flags}
  -DDEBUG_LEVEL=0 -DSS_TIMER=4
monitor_speed     = 250000
lib_deps          = ${common_stm32f1.lib_deps}
```

Is this a mistake on that file or something that's unique to my build because I don't recall needing to play with this before and I've been building v2 firmwares since the start of the year on both machines?

### Configuration Files

[Files.zip](https://github.com/MarlinFirmware/Marlin/files/5634574/Files.zip)

### Steps to Reproduce

1. Fresh install of VScode and clean out any downloaded dependant libraries from .pio folder especially `.pio\libdeps\STM32F103RC_btt_512K`
2. Complile Marlin for SKR mini E3 v1.2

**Expected behavior:**

Clean compilation

**Actual behavior:**

Fails to find SoftwareSerial.h

[image]

#### Additional Information

I am using Windows 10, the latest VScode 1.51.1, with the Auto Marlin Builder 2.1.24 and PlatformIO IDE 2.2.1

Although I know my fix works for me, I'm not sure if that was the correct place to fix it knowing how complex the whole dependency thing is, so there could be a better place to fix it which I am happy to test if you propose it.

It seems the problem is not detected unless you have a fresh install and clean download of files because removing the line again and recompiling continues to work because the previous files still exist in .pio - the above screen shot was taken after removing my fix and deleting just the "SoftwareSerialM" folder.

sjasonsmith, 2020-12-03:
That should be handled automatically. I don't know why you're hitting it and the rest of us are not. Perhaps something in your configs is preventing the automatic dependency detection from working properly.

ellensp, 2020-12-03:
I'm having lots of issue with paltformio downloading things currently.... getting http 500 errors. 
May be related.... I was just going to try again later.

Taomyn, 2020-12-03:
> That should be handled automatically. I don't know why you're hitting it and the rest of us are not. Perhaps something in your configs is preventing the automatic dependency detection from working properly.

It was a clean install of everything to compile Marlin, so not sure what it would be, but I did manually force the issue on the other machine that had been working fine and it failed there.

Also, if platformio.ini is not telling VScode that SoftwareSerialM is a dependency for my build then how would anything else in VScode detect it?

sjasonsmith, 2020-12-03:
The `extends` line should automatically use the field from the parent it points to, which should roll its way up to the dependencies in `common_stm32f1`.

Taomyn, 2020-12-03:
I think I see the problem, just need to work it out when I have a free moment

thisiskeithb, 2020-12-03:
> I'm having lots of issue with paltformio downloading things currently.... getting http 500 errors.
> May be related.... I was just going to try again later.

It’s been an issue all night for me as well. I gave up and figured I’d check in the morning.

l0l0l0, 2020-12-03:
Hello,
I have the same problem i have been trying to solve for a week, unable to compile for my SKR PRO. Can you help me.

[image]

Thanks

sjasonsmith, 2020-12-03:
@l0l0l0 There is a pretty big difference between the SKR E3 and SKR Pro with regard to software serial.

The SKR E3 pulls in an external dependency to get it.

The SKR Pro gets it from the Arduino framework it uses. If it is missing I am guessing that you are running some really old code.

l0l0l0, 2020-12-03:
Oh ok !
my code is from InsanityAutomation, but it doesn't look that old. Before softwareserial.h / cpp was in Marlin / Marlin / src / HAL / STM32 / but it's not there anymore.

I will still do some research, thanks anyway for your answer

l0l0l0, 2020-12-03:
Oh ok !
my code is from InsanityAutomation, but it doesn't look that old. 
Before softwareserial.h / .cpp was in Marlin / Marlin / src / HAL / STM32 / but it's not there anymore and i can't install it in vscode.
I will still do some research, thanks anyway for your answer

thisiskeithb, 2020-12-03:
Tested with the latest `bugfix-2.0.x` (3eddbc7):

NeoPixels aren't supported on STM32F1, so I reverted `platformio.ini` to it's stock form & only enabled `default_envs = STM32F103RC_btt_512K`.

I had to also disable the following:
- `NEOPIXEL_LED`
- `LED_CONTROL_MENU`
- `CASE_LIGHT_USE_NEOPIXEL`
- `SHOW_CUSTOM_BOOTSCREEN` and `CUSTOM_STATUS_SCREEN_IMAGE` (since those weren't included)

Marlin now builds properly:
```prolog
Linking .pio/build/STM32F103RC_btt_512K/firmware.elf
Building .pio/build/STM32F103RC_btt_512K/firmware.bin
Checking size .pio/build/STM32F103RC_btt_512K/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  63.1% (used 31032 bytes from 49152 bytes)
Flash: [=====     ]  51.3% (used 268796 bytes from 524288 bytes)
=================================== [SUCCESS] Took 28.41 seconds ===================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  SUCCESS   00:00:28.406
==================================== 1 succeeded in 00:00:28.406 ====================================

Terminal will be reused by tasks, press any key to close it.
```

thisiskeithb, 2020-12-03:
> my code is from InsanityAutomation, but it doesn't look that old.

@l0l0l0: We can't support forks here, so you'll need to post the bug report in `InsanityAutomation` repo.

closed 2020-12-03 by thisiskeithb

Taomyn, 2020-12-03:
> > NeoPixels aren't supported on STM32F1
Really? My STM32F1 based board i.e. SKR mini E3 v1.2 in my Ender 3 Pro says otherwise, and has done since the start of 2020 with nothing more than 7 standard NeoPixels connected directly to the board just as BTT designed it. This was taken after flashing it with bugfix-2.0.x from a few hours ago and correcting the platformio.ini file: https://photos.app.goo.gl/ZvFZAkXRyzVnbJ9U8

thisiskeithb, 2020-12-03:
> Really? My STM32F1 based board i.e. SKR mini E3 v1.2 in my Ender 3 Pro says otherwise, and has done since the start of 2020 with nothing more than 7 standard NeoPixels connected directly to the [...]

You're bypassing the sanity check. It does not work consistently on STM32F1 so it's disabled:
https://github.com/MarlinFirmware/Marlin/blob/3eddbc7286ce2dd5a14ceb848ef26fd1dd7a3c4c/Marlin/src/HAL/STM32F1/inc/SanityCheck.h#L42-L44