MarlinFirmware/Marlin #20364: [BUG] MKS Pins

Doyle4, 2020-12-03:
Not sure if a bug or if I’m looking in the wrong places.
If I remember correct, MKS GEN L pins use to be in the pins section under A/MEGA, been a few months since compiling a firmware for it due to upgrade.
I can no longer see them in any sections under pins, can someone point me in the right direction please.

ellensp, 2020-12-03:
Marlin/src/pins/ramps/pins_MKS_GEN_L.h

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-03 by ellensp

sjasonsmith, 2020-12-03:
The easiest way to locate the correct pins file is to open pins.h and search for your board.

Doyle4, 2020-12-03:
Thanks @ellensp for pointing me to right location.

Doyle4, 2020-12-03:
> The easiest way to locate the correct pins file is to open pins.h and search for your board.

I did and couldn't find it, which lead me to asking here. :)