MarlinFirmware/Marlin #20366: [BUG] Marlin bugfix doesnt boot, MKS Rumba32

Undercoverkarotte, 2020-12-03:
### Bug Description

Newest Marlin bugfix doesnt boot.

### Steps to Reproduce

1. [First Step]
flash firmware with VSC

**Expected behavior:**

Marlin boot up

**Actual behavior:**

Show only the bootscreen, nothing else happens.

#### Additional Information

VSC terminal output:
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/5637922/Config.files.zip)

Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
============================================= [SUCCESS] Took 174.57 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
rumba32        SUCCESS   00:02:54.566
============================================== 1 succeeded in 00:02:54.566 ==============================================

sjasonsmith, 2020-12-04:
Has this configuration worked previously?

sjasonsmith, 2020-12-04:
I can confirm the problem when using the provided configs. If I only enable the board and display it boots, albeit with slight display timing issues.

I have not yet tried to isolate which option causes it to not boot properly. There are quite a few features enabled. My best guess is that something is causing a timer conflict that isn't being detected at build time.

Undercoverkarotte, 2020-12-04:
It works fine if I copy and paste the config files to the stable branch and deactivate FAST_PWM_FAN and Z-Probe wizard.
FAN_SOFT_PWM is also deactivated in Pins file.

sjasonsmith, 2020-12-06:
Can you enable `TMC_USE_SW_SPI` to see if it allows you to boot and properly communicate with your drivers?

Undercoverkarotte, 2020-12-06:
> Can you enable `TMC_USE_SW_SPI` to see if it allows you to boot and properly communicate with your drivers?

I tried this with the newest Bugfix brach but unfortunately it doesnt change anything..

sjasonsmith, 2020-12-06:
OK, that is unfortunate. I was able to reproduce a hang but it went away with `TMC_USE_SW_SPI`.

Can you comment out all your TMC5060 drivers, to test whether the hang involves communication with them? Perhaps I am encountering something entirely different.

rhapsodyv, 2020-12-06:
@Undercoverkarotte if you can, try test disabling `SD_SUPPORT`.

sjasonsmith, 2020-12-19:
@Undercoverkarotte are you still interested in helping debug this? We would like you to try disabling SD_SUPPORT to determine if that is the conflict causing problems.

sjasonsmith, 2020-12-29:
Closing, as reporter is unresponsive.

closed 2020-12-29 by sjasonsmith