MarlinFirmware/Marlin #20373: [BUG] 3rd extruser is not turning while SDSUPPORT is enabled

kyaron, 2020-12-04:
Configurtion = 
BTT SKR1.4 Turbo
BTT TMC2209
BTT MOT-EXP
Marlin 2.x bugfix.

I am using BTT MOT-EXP in order to have a 3rd extroder. (X, Y, Z, E1, E2, E3).

While configuring the Marlin bugfix 2.x i found out that everytime I enable the SDSUPPORT, the E3 is not turning.

Tried to cross everything HW releated - cables, drivers, motors, non helped. The only way to get it back to work is commenting it.
[3rd E issue - Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5645125/3rd.E.issue.-.Configuration.zip)

### Configuration Files

Included

### Steps to Reproduce

1. Uncomment "SDSUPPORT", line 1771, Configuration.h
2. Try to extrode

**Expected behavior:**

All 3 extruders needs to turn

**Actual behavior:**

X, Y, Z, E1, E2 are turning without any issue all the time. E3 turns only if "SDSUPPORT" is disabled

rhapsodyv, 2020-12-04:
For this board, the default is the SD on LCD. So, Marlin will try to communicate with the SD using the same EXP ports that you connect MOT-EXP.
So, a `SDCARD_CONNECTION == LCD` will never work with MOT-EXP.

Can you try again, but setting the SD to ONBOARD?

`#define SDCARD_CONNECTION ONBOARD`

kyaron, 2020-12-04:
Tried, same issue

 

Thanks,

 

Yaron Kaufman

rhapsodyv, 2020-12-04:
Very weird, because with SD onboard, no PIN is shared between EXP ports and the SD.

Minims, 2020-12-05:
I have the same issue on SKR Pro with SD ONBOARD.
If I remove the microSD card from the SKR board, it works. But the SD Card can't be used.

Minims, 2020-12-06:
Additional information. I have test with TMC2209 UART and DRV8825 (which use only EXP2 wire). I have the issue in both case. Can it be detect_pin or spa mode for ONBOARD SD ?

rhapsodyv, 2020-12-06:
> Additional information. I have test with TMC2209 UART and DRV8825 (which use only EXP2 wire). I have the issue in both case. Can it be DETECT_PIN or SPI mode used for ONBOARD SD ?

Not according with the schematics. SD detect pin is `P0_27`, but exp ports don't use this pin:
```cpp
/*               _____                                      _____
 *           NC | · · | GND                             NC | · · | GND
 *           NC | · · | 1.31 (M1EN)            (M2EN) 1.23 | · · | 1.22 (M3EN)
 * (M1STP) 0.18 | · ·   3.25 (M1DIR)           (M1RX) 1.21 | · ·   1.20 (M1DIAG)
 * (M2DIR) 0.16 | · · | 3.26 (M2STP)           (M2RX) 1.19 | · · | 1.18 (M2DIAG)
 * (M3DIR) 0.15 | · · | 0.17 (M3STP)           (M3RX) 0.28 | · · | 1.30 (M3DIAG)
 *               -----                                      -----
 *               EXP2                                       EXP1
 */
```

It's a mysterious. I ordered one mot-ext to test it.

ellensp, 2020-12-06:
I'm thinking the SPI ports that would be used to talk to the SD card is being setup accidentally

rhapsodyv, 2020-12-06:
> I'm thinking the SPI ports that would be used to talk to the SD card is being setup accidentally

But the current HW SPI code for LPC just init the SPI port if it's used.... Yet it would not explain why enabling/disabling the onboard sd would influence in anything, as it use different SPI port that the EXP slots... :-/ I can't see any shared PIN.

ellensp, 2020-12-06:
Well, I also have one of these on order (for a while now)..  all going well it should arrive in about a week. then can try some stuff.

Minims, 2021-01-06:
Hello, problem still exists, just keeping this issue open.

ellensp, 2021-05-01:
perhaps this is the issue https://github.com/MarlinFirmware/Marlin/pull/21755

closed 2021-07-13 by github-actions[bot]