MarlinFirmware/Marlin #20375: [BUG] The temperature drops to the zero immediately before main print

o5, 2020-12-05:
### Bug Description

The temperature of hot-end and bad drops to the zero after pre-print process finished. Happening randomly, with print over OctoPrint also directly from SD card.

### Configuration Files

[Configuration.h](https://github.com/diy4fun/marlin-configurations/blob/e8eb90/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration.h)
[Configuration_adv.h](https://github.com/diy4fun/marlin-configurations/blob/e8eb90/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration_adv.h)
[Marlin based on `bugfix-2.0.x` branch](https://github.com/diy4fun/marlin/commits/bugfix-2.0.x)
[Starting G-CODE](https://gist.github.com/o5/3f6e5a344e9cfb3b2409cb0f120a354e)

### Steps to Reproduce

1. Starting G-CODE from Slicer works normally (heat-up, auto-calibration)
2. When first con contour of object is created the temperatures (hot-end, bed) are changed to zero
3. `G0/1` commands continue (printer is moving) but when temperature down extruder stop to moving (during protection I think)

**Expected behavior:**

Every print should not have this behaviour.

**Actual behavior:**

Unfortunately it doing randomly.

sjasonsmith, 2020-12-05:
This sounds most likely to be a wiring issue to me. It is possible for wires to be broken but only lose connection randomly.

One idea is for you to heat bed/extruder, then manually move your bed/carriage slowly, to see if your temperatures are lost at any point in the travel.

Minims, 2020-12-06:
Does it work if you comment this ?  '
```
#define HOTEND_IDLE_TIMEOUT
```

o5, 2020-12-07:
@sjasonsmith Thanks for tip! I changed to V6 hotend (but two months ago), so it could be. But what wires you mean? Termistors for hotend and bed? But it's very weird because it's happened (if happened) always after pre-print process finished.

@Minims Thanks for tip! There [should be always](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/hotend_idle.cpp#L77) message on the screen when timeout also this timeout should be disabled when moving, or not?

@rhapsodyv Could you tell me what more data I should attach?

rhapsodyv, 2020-12-07:
@o5 I added "needs more data" because we are waiting you to check about wiring, as Jason asked. It's just for our internal control, that we asked to user to do something.

sjasonsmith, 2020-12-07:
After seeing @Minims comment this probably is just a hotend timeout during G29. Hotend timeout is probably based on extruder movement, not the XYZ movement used by G29.

thisiskeithb, 2020-12-07:
> After seeing @Minims comment this probably is just a hotend timeout during G29. Hotend timeout is probably based on extruder movement, not the XYZ movement used by G29.

For long-running G29 (especially UBL), increasing default `HOTEND_IDLE_TIMEOUT` will help.

Would it make sense for long-running bed leveling operations to reset the timer in `HOTEND_IDLE_TIMEOUT`?

Minims, 2020-12-07:
> Would it make sense for long-running bed leveling operations to reset the timer in `HOTEND_IDLE_TIMEOUT`?
 Something like `RESET_ HOTEND_IDLE_TIMEOUT_AFTER_G29` seems to be a good ideal. In particular for UBL or ABL 7x7 with BLTOUCH for example.

o5, 2020-12-08:
@Minims Your tip makes sense to me, when I looked to my `HOTEND_IDLE_TIMEOUT` (which is 5min) because I recently increased bed temperature and it really takes more time to heat-up so pre-print process is more time consuming now. 

1. I don't understand why I don't see the info message about timeout on screen when this problems happens. I know very well this message from manual heat-up but in this case no message on the screen.
1. And second thing is why steppers (except extruder - I think thanks to low temperature) continue. What about feature to stop it when timeout appears? 
1. I like idea with `RESET_HOTEND_IDLE_TIMEOUT_AFTER_G29`

rhapsodyv, 2020-12-08:
Hi @o5. So `HOTEND_IDLE_TIMEOUT` was the reason for the temp issue?

> I don't understand why I don't see the info message about timeout on screen when this problems happens. I know very well this message from manual heat-up but in this case no message on the screen.

When it happens in the middle of another ongoing process, one message will overwrite the other... :-/ Currently, the only way to keep a message permanent on the screen is if we halt it entirely. 

> And second thing is why steppers (except extruder - I think thanks to low temperature) continue. What about feature to stop all steppers when timeout appears?
> I like idea with RESET_HOTEND_IDLE_TIMEOUT_AFTER_G29

Who will open a FR for this two ideas? `RESET_HOTEND_IDLE_TIMEOUT_AFTER_G29` and `HOTEND_IDLE_TIMEOUT "stop all"`

o5, 2020-12-09:
> So HOTEND_IDLE_TIMEOUT was the reason for the temp issue?

It looks like it. Two prints with increased `HOTEND_IDLE_TIMEOUT` and without issue. I'll try more and give a feedback.

thisiskeithb, 2020-12-09:
Closing since this feature is working as designed (but may not be desired for long probing actions).

Please open a [Feature Request](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=T%3A+Feature+Request&template=feature_request.md&title=%5BFR%5D+%28feature+request+title%29) to enhance `HOTEND_IDLE_TIMEOUT` to reset during long-running commands like `G29`.

closed 2020-12-09 by thisiskeithb

o5, 2020-12-10:
@thisiskeithb If I'm thinking about what to write into feature request I'm a little bit confused. My ABL takes about 2minutes max, so it will be probably connected to my increasing  temperature of bed temp or combination both of them.

o5, 2021-01-07:
I can confirm increasing of `HOTEND_IDLE_TIMEOUT` (`5*60` => `10*60`) really resolved the issue.