MarlinFirmware/Marlin #20377: [BUG] (MKS Robin V1.1R TFT - blank screen with TFT_COLOR_UI enabled)

mattdog01, 2020-12-05:
### Bug Description

Using the latest (downloaded 20201205 Marlin-bugfix-2.0.7.2) bugfix.  
My screen is completely blank if I uncomment #define TFT_COLOR_UI but it works just fine if I uncomment #define TFT_CLASSIC_UI. Touchscreen works in both configs but without visual reference in the blank screen.
I did minimal changes to the config files to exclude other factors. Only changed physical things in the configuration.h file (attached) no changes to configuation_adv.h. 

/**

TFT UI - User Interface Selection. Enable one of the following options:
TFT_CLASSIC_UI - Emulated DOGM - 128x64 Upscaled
TFT_COLOR_UI - Marlin Default Menus, Touch Friendly, using full TFT capabilities
TFT_LVGL_UI - A Modern UI using LVGL
For LVGL_UI also copy the 'assets' folder from the build directory to the
root of your SD card, together with the compiled firmware.
*/
//#define TFT_CLASSIC_UI
#define TFT_COLOR_UI
//#define TFT_LVGL_UI

### Configuration Files

[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/5647413/Config.files.zip)

### Steps to Reproduce

1. I first tried the build in the mks_robin environment uncommenting #define TFT_CLASSIC_UI
2. Booted up the machine with the SD card containing the new bin file.
3. Worked perfectly.  Classic UI visible and touch works perfectly.
4. Cleaned the build area and tried the build again but with only #define TFT_COLOR_UI uncommented.
5. Booted up the machine with the SD card containing the new bin file.
6. Screen is blank once the upload is complete. Touch works but without the visual reference.

**Expected behavior:**

I expected the TFT_COLOR_UI to work and be visible like it was for earlier builds.

**Actual behavior:**

The screen is completely blank no light at all. **The text can be seen if a strong light is held up to the screen.**
(Suspect backlight pin is not defined correctly for the board - ¯\_(ツ)_/¯ )

#### Additional Information

The board
   MKS Robin v2.3

The screen
   MKS TFT v1.1R

The machine
   coreXY printer
   running on 24volts

The compiler
   PlatformIO
   mks_robin build environment specified in the platformio.ini

Last fully functional bug_fix was downloaded 2020-09-09

mattdog01, 2020-12-05:
What I can find and it all looks correct.  The drawings specify PG11 for the FSMC_LIGHT

pin.h
#elif MB(MKS_ROBIN)
  #include "stm32f1/pins_MKS_ROBIN.h"           // STM32F1                                env:mks_robin env:mks_robin_stm32

pins_MKS_ROBIN.h
  //#define LCD_RESET_PIN                      PF6
  #define LCD_BACKLIGHT_PIN                PG11
  #define FSMC_CS_PIN                           TFT_CS_PIN
  #define FSMC_RS_PIN                           TFT_RS_PIN

  #define LCD_USE_DMA_FSMC                // Use DMA transfers to send data to the TFT
  #define FSMC_DMA_DEV                       DMA2
  #define FSMC_DMA_CHANNEL              DMA_CH5
#elif HAS_GRAPHICAL_TFT
  #define TFT_RESET_PIN                          PF6
  #define TFT_BACKLIGHT_PIN                 PG11
#endif

rhapsodyv, 2020-12-07:
The pins file were outdated, after some TFT changes. 

Fixed here: #20404 . Can you test it?

mattdog01, 2020-12-08:
@rhapsodyv  Hello Victor thank you for your work.  I downloaded a fresh copy of bugfix an 1hr ago and built the bin file and it is the same issue except I have also lost touch screen function.  The display is visible in classic mode but not in colour UI mode.  Touch screen doesn't work in either mode.  
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5660922/Configuration.zip)

mattdog01, 2020-12-08:
oh i forgot to mention I'm using the mks_robin build environment not the mks_robin_stm32 environment.  Is that how you wanted it tested?

rhapsodyv, 2020-12-08:
My fix was not merged on bugfix yet. I'm waiting you test it, before merging in bugfix

rhapsodyv, 2020-12-09:
Fixed and merged.

PR #20404 (2020-12-09): MKS Robin TFT pins fixes for STM32 and STM32F1
### Description

Update the TFT pins for the MKS Robin board after the TFT refactoring.

Fixed a few more issues on this board, for STM32F1:

 - There's no SD_DETECT_PIN on MKS Robin V2.3 (Matt board). Only on > V2.4.
 - Fixed touch on STM32F1 too

### Benefits

Fix #20377
Fix #19274 

### Related Issues 

#15474
closed 2020-12-09 by rhapsodyv