MarlinFirmware/Marlin #20382: Problems with G2/G3

matbhe, 2020-12-05:
### Bug Description
The following G2 - Test on the `bugfix-2.0.x` branch failed (no movement):
G21
G90
G0 X-3.50 Y0.00
G2 I3.50 J0

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5647969/Configuration.zip)
Done some modifications for my CNC setup with TMC5160s, sensorless homing and TFT24.

### Steps to Reproduce
1. Run G-Code
2. Check if machine is moving

**Expected behavior:**
The machine should machine/draw a circle.

**Actual behavior:**
The machine does not move.

ellensp, 2020-12-06:
Usage
`G2 [E<pos>] I<offset> J<offset> [P<rate>] R<radius> [X<pos>] [Y<pos>] [Z<pos>]`

I,J and R are required.

matbhe, 2020-12-06:
@ellensp I think you are mixing the two forms (IJ-Form and R-Form).
According to the manual at https://marlinfw.org/docs/gcode/G002-G003.html the I,J and R should not be used at the same time: "**Mixing I or J with R will throw an error.**"...

ellensp, 2020-12-06:
You are correct. I am mistaken.

yysh12, 2020-12-06:
This sounds like a support issue not a bug. You should go to the discord channel for support issues. If they cannot solve it then file a bug report.

sjasonsmith, 2020-12-06:
There have been changes to the arc code recently. We should go ahead and leave this open until we verify whether it is or is not a problem.

yysh12, 2020-12-06:
Ok, in that case, @matbhe try to run the 2.7.2 branch to see if the problem is still present.

matbhe, 2020-12-06:
Ok. I've tested it with the version tagged with 2.0.7.2 (cannot find 2.7.2) - same result.
I've done some debugging and the problem seems to be located somewhere around the following statement in G2_G3.cpp:

 switch (((angular_travel < 0) << 1) | clockwise) {
      case 1: angular_travel -= RADIANS(360); break; // Positive but CW? Reverse direction.
      case 2: angular_travel += RADIANS(360); break; // Negative but CCW? Reverse direction.
 }

I've inserted the debug statement _SERIAL_ECHOPAIR("at: ", angular_travel);_ before. If the output is "**at: 0.00**" then everything is fine. But if the output is "**at: -0.00**" then it does not work.

yysh12, 2020-12-06:
Yeah sorry meant 2.0.7.2. And that's interesting. The code you mentioned is not present in 2.0.7.2. It was just introduced into the bugfix. Can you tell me specifically where you added that debug statement (preferably just paste into a code block)?

matbhe, 2020-12-06:
_char buff[10];
    snprintf (buff, sizeof(buff), "%f", angular_travel);
    SERIAL_ECHOPAIR("at1: ", buff);_

    switch (((angular_travel < 0) << 1) | clockwise) {
      case 1: angular_travel -= RADIANS(360); break; // Positive but CW? Reverse direction.
      case 2: angular_travel += RADIANS(360); break; // Negative but CCW? Reverse direction.
    }

Without Homing (G28) the output is 0.000001 and it works. With homing before the execution the output is -0.000003 and it does not work ...

yysh12, 2020-12-06:
Does your if statement look like this:
`if ((!angular_travel || NEAR_ZERO(angular_travel)) && NEAR(current_position[p_axis], cart[p_axis]) && NEAR(current_position[q_axis], cart[q_axis])) {`

or like this:

`if (!angular_travel || (NEAR_ZERO(angular_travel) && NEAR(current_position[p_axis], cart[p_axis]) && NEAR(current_position[q_axis], cart[q_axis]))) {`
?

matbhe, 2020-12-06:
I'm on branch bugfix-2.0.x and it looks like the second one. Beside the G28 statement I'm also executing G92 X0 Y0 Z0 before running the code mentioned in the description...

yysh12, 2020-12-06:
Okay, this makes more sense now. So this was a problem with 2.0.7.2 as well and the code was revised to fix this. But looks like something is still falling through. I suspect it may be due to a difference in precision. So do this:
I need you to use that same debug statement multiple times at the following points marked with //PROBE

```
// Angle of rotation between position and target from the circle center.
  float angular_travel = ATAN2(rvec.a * rt_Y - rvec.b * rt_X, rvec.a * rt_X + rvec.b * rt_Y);

//PROBE angular_travel
//PROBE current_position[p_axis] 
//PROBE cart[p_axis]
//PROBE current_position[q_axis]
//PROBE cart[q_axis]

  #ifdef MIN_ARC_SEGMENTS
    uint16_t min_segments = MIN_ARC_SEGMENTS;
  #else
    constexpr uint16_t min_segments = 1;
  #endif

  // Do a full circle if angular rotation is near 0 and the target is current position
  if (!angular_travel || (NEAR_ZERO(angular_travel) && NEAR(current_position[p_axis], cart[p_axis]) && NEAR(current_position[q_axis], cart[q_axis]))) {
    // Preserve direction for circles
    angular_travel = clockwise ? -RADIANS(360) : RADIANS(360);
  }
  else {
    // Make sure angular travel over 180 degrees goes the other way around.
    switch (((angular_travel < 0) << 1) | clockwise) {
      case 1: angular_travel -= RADIANS(360); break; // Positive but CW? Reverse direction.
      case 2: angular_travel += RADIANS(360); break; // Negative but CCW? Reverse direction.
    }
    #ifdef MIN_ARC_SEGMENTS
      min_segments = CEIL(min_segments * ABS(angular_travel) / RADIANS(360));
      NOLESS(min_segments, 1U);
    #endif
  }
```

yysh12, 2020-12-06:
@sjasonsmith, pending the results from @matbhe:
Go ahead and mark this as a bug. I think I know what the problem probably is:
the near_zero macro tests for a value between -0.000001 and 0.000001, but the user is getting a result from atan2 that is -0.000003 which falls outside that range and is getting treated as a non-circle (the if statement is for circles and the else statement is for non-circles). Ideally any full circle motion should be handled within the if statement and not within the else statement. Assuming this isn't some number computation precision issue depending on the platform (i.e. 8-bit vs 32-bit or something), the solution would be to expand that range to something larger (i.e. -0.000005 and 0.000005), but that is something that falls out of my field of knowledge. I don't know what precision would work best across all systems or if that is even something that can vary across platforms. We don't want it to be too large that non-circles get caught, but we don't want it to be too small that circles outside the range get missed. 
But once again we should get the results from @matbhe before we can be sure.

matbhe, 2020-12-06:
Here the results:

**Test with G28 and G92 X0 Y0 Z0 (not working):**
angular_travel: 0.000000
current_position[p_axis]: 46.650002
cart[p_axis]: 46.650002
current_position[q_axis]: 48.980000
cart[q_axis]: 48.980000

angular_travel: -0.000003
current_position[p_axis]: 46.650002
cart[p_axis]: 46.650002
current_position[q_axis]: 256.97998
cart[q_axis]: 256.97998

**Test without G28/G29 (working):**
angular_travel: 0.000000
current_position[p_axis]: -3.350000
cart[p_axis]: -3.350000
current_position[q_axis]: -1.020000
cart[q_axis]: -1.020000

angular_travel: 0.000001
current_position[p_axis]: -3.350000
cart[p_axis]: -3.350000
current_position[q_axis]: 206.97999
cart[q_axis]: 206.97999

Hope this helps!

yysh12, 2020-12-06:
Yep. This confirms it. The near_zero macro needs to be revised. The range is too small for this specific user and quite possibly others as well (once again not sure if this is something that can vary across platforms or processor architectures). @matbhe, please change the Near_zero macro to use a larger range (.i.e. maybe +- 0.000005) and see if the problem resolves.

matbhe, 2020-12-06:
@yysh12 Yes... this solved the problem - Thank you! I will use this fix as temporary solution...

yysh12, 2020-12-06:
Great! Now we just need someone who actually knows what they are doing to make a decision on what the best range for all users should be! @sjasonsmith Please advise!

sjasonsmith, 2020-12-06:
I don't have a recommendation. I don't have know how to use G2/G3 😁

yysh12, 2020-12-11:
@matbhe 
Can you please test the proposed fix to see if your problem resolves? Also, if possible, try additional arcs to ensure no unwanted side effects occur (specifically looking for scenarios where you didn't request a full circle yet a full circle still occurred). I essentially deleted the requirement that angular travel be near zero. Circle determination is now based directly from position and target vectors.

yysh12, 2020-12-11:
There are now two alternate proposals. Try the first especially, but if you can try the second one as well.

matbhe, 2020-12-14:
@yysh12 I've tested all proposals and could not find any issues 😃 ...

yysh12, 2020-12-14:
Great! Then it is definitely a possible alternative to increasing range of Near_zero macro. Thanks for testing it out!

sjasonsmith, 2020-12-24:
@matbhe the change from @yysh12 has been merged into the bugfix-2.0.x branch. Can you verify things are now working properly, and close this issue if all is resolved?

closed 2021-01-31 by github-actions[bot]