MarlinFirmware/Marlin #20386: [FR] Probe offset wizard: Heat nozzle, retract filament

morganchristiansson, 2020-12-06:
### Description

Make z probe wizard heat nozzle to configurable temperature. Measuring a cold nozzle will result in incorrect offset result.

Also retracting filament so it doesn't dribble out the nozzle while calibrating.

### Feature Workflow

1. When starting z probe offset wizard
2. Ask user for desired temperature
3. Display progress while waiting for nozzle
4. Retract filament so it doesn't dribble during calibration
5. run z probe offset wizard as usual
6. when exiting wizard undo the filament retraction and cool down the nozzle

Also fantastic work on adding the z probe offset wizard feature!

firepower9966, 2020-12-06:
should have choice for Probe Offset Wizard same as Build Mesh Cold or Build Mesh PLA or Build Mesh ABS (The two default preset temperature defines)

Probe Offset Wizzard Cold
Probe Offset Wizzard PLA
Probe Offset Wizzard ABS

hamster65, 2020-12-06:
+1. 
I Proposed this as well because it is also relevant for my bltouch which has a very low z offset for mechnial reasons. Blobs on the nozzle cause the nozzle to make contact with the bed before the bltouch triggers.
I was told to just add heating to my start code before G28.