MarlinFirmware/Marlin #20387: [BUG]  "One or more timer conflict detected" when trying to compile using mks_robin_nano35_stm32

th0m4sek, 2020-12-06:
### Bug Description

```
`Marlin/src/HAL/STM32/timers.cpp:281:41: error: static assertion failed: One or more timer conflict detected
  281 | static_assert(verify_no_duplicate_timers(), "One or more timer conflict detected");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~^~`
```

same configuration working fine when compiling with mks_robin_nano35
using tmc2209 with uarts connected to one pin.

### Configuration Files

[Marlinconf.zip](https://github.com/MarlinFirmware/Marlin/files/5648554/Marlinconf.zip)

### Steps to Reproduce

compile attached files

**Expected behavior:**

posibility to compile with mks_robin_nano35

**Actual behavior:**

th0m4sek, 2020-12-08:
Solution from [PIO env for MKS Nano boards in STM32 #19905] (https://github.com/MarlinFirmware/Marlin/pull/19905)
`#if defined(ARDUINO_ARCH_STM32) && !defined(STM32GENERIC)
  // Variant default conflicts with Servo timer
  #define TEMP_TIMER 5
#endif`
worked for my MKS Robin Nano V1.2
Maybe this should be merged?

rhapsodyv, 2020-12-08:
@sjasonsmith we can merge it? or it's just a temporary hack?

sjasonsmith, 2020-12-08:
I would like to try to find a more flexible way to solve it. As more boards move to the STM32 HAL this will continue to be a problem, and it would be nice to not have to manually define the timer for each board. Not every board should use the same timers for this, due to other conflicts.

sjasonsmith, 2020-12-20:
I've updated the provided configs to reproduce the problem using the current state of bugfix-2.0.x.
I tried to preserve all the original settings, but no promises. My goal is just to reproduce the compilation issue.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5720601/Configuration.zip)

th0m4sek, 2020-12-20:
updated conf on current bugfix is acting as before - can be compiled with mks_robin_nano but not with mks_robin_nano_stm32

sjasonsmith, 2020-12-20:
Sorry, I wasn't suggesting that I've fixed it, I just updated them to provide failing configs that work with the current bugfix.

I think I have a more flexible solution semi-ready, but I haven't yet tried it with any hardware. I'm about to post it as a Draft pull request to see if it passes our build tests, and to solicit feedback.

th0m4sek, 2020-12-20:
i was not sure what you wanted to communicate but i wanted to show that i'm ready to help at least with testing ;)

sjasonsmith, 2020-12-20:
@th0m4sek I appreciate the willingness to help. I just posted that draft. I'll tag you again when it is ready for you to try out. I'd like to test it with some of my own hardware first.

sjasonsmith, 2020-12-22:
This conflict should now be resolved, along with several other boards with similar issues.
I'm closing this, but please let us know if you continue to see issues with the latest changes from `bugfix-2.0.x`.

PR #20545 (2020-12-22): Override STM32 timer assignments to avoid conflicts
### Description

Several STM32 boards had timer conflicts when enabling SPEAKER, BLTOUCH, and TMC2208 drivers with SoftwareSerial.
Override these timer settings in the pins file.
For one board the SoftwareSerial timer number was added in platformio.ini, prevent conflicting with other timers in variant.h.

I have not tested these on hardware, but I did review the PeripheralPins files to check for conflicts with fans.

### Benefits

Fewer complaints about timer conflicts when enabling extra features on boards.

### Configurations

See reference [... cut]
closed 2020-12-22 by sjasonsmith