MarlinFirmware/Marlin #20388: [BUG] UBL calculates wrong points that can't be reached

gbkwiatt, 2020-12-06:
I am running on latest bugfix 2.0.x

I am trying to figure out correct settings for UBL, and why I can't get all points probed. 
So currently it doesn't matter if I have 100 points or 49 points, it finishes before it get's to all points. 

My issue is, why Marlin allows points that can't be reached ( I am assuming that's why it can finish) 

```Cpp
// The size of the print bed
#define X_BED_SIZE 288
#define Y_BED_SIZE 250

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 299 // it can be reached by probe
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 285

#define MESH_INSET 3 
#define PROBING_MARGIN 0

#define NOZZLE_TO_PROBE_OFFSET { -34, -1, -2.03 }
```

That's my current settings for UBL. 
I believe, it would be nice, to have some check on compiling, that at least Warns about unreachable points in UBL. 
But also UBL Marlin should be smart enough to calculate points only based on the ones that it can reach.
Or at lest some documentation on how to calculate those settings.

ellensp, 2020-12-06:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2020-12-06:
The purpose of allowing travels beyond the bed area is because of endstops positions and a few other usages, like to change filament out of the bed area.
I don't think we should give a warn or error about that, because it will just give false-negatives for a lot of common setups.

closed 2020-12-06 by rhapsodyv

gbkwiatt, 2020-12-06:
I understand that, but why even given that data UBL still calculate points that can't be reached?
So it works again the limits set, because UBL has it's point calculated, and when it wasn't to go to it, Maril limits says no and UBL ends. 
Shouldn't UBL have points always inside allowed travels?

sjasonsmith, 2020-12-06:
If you mean it calculates points the _probe_ cannot reach, this is by design. UBL covers the entire bed, so some points have to be filled in automatically (with `G29 P3`, I think?) or probed manually.

If you reduce the mesh size to only include the points your probe can reach, then you will not have bed leveling across your entire printable area.

rhapsodyv, 2020-12-06:
> I understand that, but why even given that data UBL still calculate points that can't be reached?
> So it works again the limits set, because UBL has it's point calculated, and when it wasn't to go to it, Maril limits says no and UBL ends.
> Shouldn't UBL have points always inside allowed travels?

I never saw it going where it shouldn't. But if if think that found a bug, please follow the issue template as requested.

gbkwiatt, 2020-12-07:
I think I actually just clicked and realised how it works, and it all makes sense now... Sorry for messing around.