MarlinFirmware/Marlin #20390: SKR 1.4 with TMC2208 halts after moving axes back and forth with homing

ebraiman, 2020-12-06:


ebraiman, 2020-12-06:
When homing after moving x,y,z axes, Printer halts.

Hardware: 
BTT TMC2208 v3.0
BTT SKR 1.4

Software:
Marlin 2.0.7.2
Pronterface
Cura 4.8 (Slic3r)

Expected result:
No homing issues after moving axes.

Actual result:

Driver registers:
		X	0xC0:0B:00:00
		Y	0xC0:0B:00:00
		Z	0xC0:0B:00:00
		E	0xC0:0B:00:00
		E1	0xC0:0B:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
X:37/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:35/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:39/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:35/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:39/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:13/11|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:13/11|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:81/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:72/23|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:70/23|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:82/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:72/23|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:81/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:82/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:81/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:39/11|TI|	Y:82/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:72/23|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:82/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:81/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:82/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:83/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:72/23|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:73/23|TI|	Z:34/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:39/11|TI|	Y:73/23|TI|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:38/11|TI|	Y:72/23|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
[... 33 lines cut ...]
X:80/23|T|	Y:39/11|TI|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:80/23|T|	Y:40/11|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:74/23|T|	Y:39/11|TI|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:72/23|TI|	Y:81/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
echo:busy: processing
X:71/23|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:73/23|TI|	Y:82/23|T|	Z:39/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:72/23|TI|	Y:82/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:73/23|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
echo:busy: processing
X:72/23|TI|	Y:82/23|T|	Z:35/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:67/21|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:61/19|TI|	Y:83/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:52/16|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
echo:busy: processing
X:47/14|TI|	Y:82/23|T|	Z:37/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:40/12|TI|	Y:82/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:39/11|TI|	Y:83/23|T|	Z:38/11|TI|	E:13/11|TI|	E1:13/11|TI|
X:37/11|TI|	Y:82/23|T|	Z:35/11|TI|	E:13/11|TI|	E1:13/11|TI|
echo:busy: processing
X:38/11|TI|	Y:82/23|T|	Z:36/11|TI|	E:13/11|TI|	E1:13/11|TI|
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/5649725/Configuration.files.zip)

ebraiman, 2020-12-06:
Also observed a change in distance for length of axes.  Stepping seemed to change from 230 as a distance of 230, to about 115 in distance.

sjasonsmith, 2020-12-07:
It sounds like you have the incorrect microstepping programmed into your drivers.

First, make sure you are not powering over usb before applying full board power. This will prevent drivers from initializing properly.

Second, enable TMC_DEBUG, and use M122 to observe the number of micro steps programmed into each driver.

sjasonsmith, 2020-12-07:
When you try to home does it move toward your endstops, but halt before reaching the endstop?

ebraiman, 2020-12-07:
Before endstop.
I moved the USB jumper just now.
Also I think there is a command issue in gcode for Cura 4.8 for tool change.  It's a single printhead and I might have configured Cura incorrectly.
I check tomorrow the changes and try in Slic3r.

closed 2021-01-14 by github-actions[bot]