MarlinFirmware/Marlin #20391: [FR] TMC_HOME_PHASE adjustable values from the menu

yodor, 2020-12-06:
### Description

Hello, thanks for making the 'TMC Homing stepper phase' feature available. 
I started to use it recently but from time to time it triggers the warning like "Selected home phase too close to endstop trigger phase. Pick a different phase for". And this means edit values in source and re-upload the firmware again. 
As there is already a submenu for TMC Drivers, (Configuration->Advanced Settings->TMC Drivers) would it be possible for an additional submenu "Phase Adjust/Home Phase" where user can pick up the values for each axis by using the LCD and the rotary switch.
 
### Feature Workflow

1. Go to Configuration->Advanced Settings->TMC Drivers
2. Select 'Home Phase' submenu
3. Select one of  (X.Y.Z axis) and then adjust the current homing phase value using the LCD rotary switch. (as it is for selecting the power for the drivers)

qwewer0, 2020-12-06:
While it is a good idea to have on the fly (gcode and/or menu item) way to modify the TMC Homing stepper phase values, but I'm not sure if the increased (I assume) firmware size would worth it, as for 2-3 axes it isn't too hard to get the right phase with a few firmware flashes.

yodor, 2020-12-07:
Well, for deltas are for sure 3 :). But also some of the newer 32-bit boards need sd-card to update the fw. Also for new builders / new machines this might need more than few times. 
On the other side 2.0 branch already can not fit into some of the 8 bit boards so I'm not sure how much size of fw is issue anymore. I would not mind to assist in preping a pull-request, just thought some more experienced with marling code would do it faster.

Portzal, 2020-12-08:
I have been meaning to ask a question on this option. I have it enabled on 2 printers, (cartesian and corexy) and I have noticed on occasion I get the warning, but not every home. I have adjusted the numbers, but still get intermittent warnings.

qwewer0, 2020-12-08:
@Portzal I think that could mean a few things, like unreliable endstop switches, endstop noise, not properly tightened belts, maybe skipped steps, or more.

Portzal, 2020-12-08:
@qwewer0 I feel like my experience relates to the original request, could it be a bug as opposed to needing to be adjusted by g code? I have sensor less homing enabled on both printer, belts are tensioned correctly and I have no indication of missed steps on either printer.

qwewer0, 2020-12-08:
@Portzal Sensor less homing is the "unreliable endstop switch"

jaymemaurice, 2022-08-27:
If not a menu, can we at least get an option to home to the nearest full stop?
It seems like a gcode or menu here would be far better use of resources than RGB LED control