MarlinFirmware/Marlin #20392: compile on Platformio for RADDS

Abysslabs, 2020-12-07:
I can compile on Platformio any marlin version to any board except for Arduino DUE+RADDS.
when I change on platformio.ini the "default_envs = DUE" I get the follow errors:

> Executing task: C:\Users\AbyssLabs\.platformio\penv\Scripts\pio.exe run <

Processing DUE (platform: atmelsam; framework: arduino; board: due)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 143, in cli
    is_test_running,
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "c:\users\abysslabs\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 59, in run
    self.configure_default_packages(options, targets)
  File "C:\Users\AbyssLabs\.platformio\platforms\atmelsam\platform.py", line 32, in configure_default_packages
    for name, opts in self.packages.items():
RuntimeError: dictionary changed size during iteration

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "C:\Users\AbyssLabs\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

rhapsodyv, 2020-12-08:
Looks like it's a problem in your platformio installation. Did you try reinstall or update PIO?

sjasonsmith, 2020-12-09:
I am able to build for RADDS just fine inside PlatformIO on Windows.

I agree that this is most likely a PlatformIO installation issue, and does not represent a bug in Marlin.

I suggest you delete your `c:\users\abysslabs\.platformio` directory. This will cause PlatformIO to re-create all of its environment and packages, which will likely resolve the issue.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-09 by sjasonsmith

Abysslabs, 2020-12-11:
@sjasonsmith 
Thanks, deleting c:\users\abysslabs\.platformio solved the problem.

@rhapsodyv 
That was the first thing that I have tested.

And SORRY late reply, I was away from home.