MarlinFirmware/Marlin #20394: [BUG] (btt mot exp & lcd setup is possibly wrong)

jasonfox7, 2020-12-07:
### Bug Description

It would appear the code of Marlin is set up to not allow the use of a wired lcd and the big tree tech MOT exp board at the same time. The btt mot exp board can be used in step/dir mode with a single cable to exp2, and a single cable lcd12864 can be connected to exp1. Is there an underlying communications bus conflict that necessitates that the two be set as mutually exclusive, or is this a design oversight as the other two communication modes of the btt mot exp require two cables?

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

Not required. 

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. View related code in the pin file for the btt skr 1.4 board. 
2. Mutually exclusive logic exists to prevent using lcd and exp board at the same time. 

**Expected behavior:**

An lcd connected to exp1 such as the cr10 display allowed to work with the btt exp board connected to exp2.

**Actual behavior:**

Exclusionary logic prevents building. 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-12-07:
Its not a combination I thought of when adding btt mot exp to marlin.

jasonfox7, 2020-12-07:
> Its not a combination I thought of when adding btt mot exp to marlin.

Thank you for the prompt reply!

Okay. So it’s theoretically possible then. I wasn’t sure how communication between exp1 and dxp2 and the mcu was handled but I assumed it was just a set of gpio pins routed to headers. 

So it’s possible then I take it? I just need to adjust the logic to allow for said combination on my end.

ellensp, 2020-12-07:
Its easy to check for a cr10 display, its rather more difficult to check that new steppers are in step/dir mode.
could be setup as e2, e3, e4 X2, Y2, Z2, Z3 or Z4 etc....

jasonfox7, 2020-12-07:
> Its easy to check for a cr10 display, its rather more difficult to check that new steppers are in step/dir mode.
> could be setup as e2, e3, e4 X2, Y2, Z2, Z3 or Z4 etc....

Alright. That makes sense, easy enough to retain the logic and add an exclusion is cr10 is defined. 

On a side note, I read in the firmware Marlin auto assigns the exp mot steppers, is it possible to specify them manually? I’m going to use the board for z2, z3

ellensp, 2020-12-07:
The default it to define the steppers as E2,E3 and E4
Then Marlin will re allocate them where needed.
Eg if you have extruders set to 2, that means marlin can used E2,E3 and E4 for anything else you defined.
ie if you have number of z 3  it will automatically reallocate E2 to Z2 and E3 to Z3  etc

Or you can edit he pins files

jasonfox7, 2020-12-07:
> The default it to define the steppers as E2,E3 and E4
> Then Marlin will re allocate them where needed.
> Eg if you have extruders set to 2, that means marlin can used E2,E3 and E4 for anything else you [...]

Intuitive solution, thank you for the explanation.

Will a futu

ellensp, 2020-12-07:
It will take some time...

ellensp, 2020-12-07:
E3_ENABLE_PIN                    EXPA1_03_PIN   ie is normally on exp1 
E4_ENABLE_PIN                    EXPA1_04_PIN   ie is normally on exp1
You have to jumper the M1EN to M2EN and M3EN and change the above defines.

jasonfox7, 2020-12-07:
> It will take some time...

Should I find time to make the relevant changes I’ll post them here. Much appreciated for the insight; where it’s not my software and I’m relatively unfamiliar with the underlying architecture used in 3d printers I wasn’t sure if it was in the realm of possibilities. 

Thank you for the clarifications, and your timeliness.

ellensp, 2020-12-07:
Try this
[pins_BTT_SKR_common.zip](https://github.com/MarlinFirmware/Marlin/files/5650593/pins_BTT_SKR_common.zip)

ellensp, 2020-12-07:
I have created a PR to support this mode. 
Instead of checking everything I went the other way. If the CR10 is enabled with a BTT_EXP_MOT, Marlin sets up the BTT EXP MOT in EXP2 only mode.

jasonfox7, 2020-12-07:
Rock on. Thanks for your help!

closed 2020-12-07 by jasonfox7