MarlinFirmware/Marlin #20397: [DRAFT] [BUG] `TMCMarlin::set_pwm_thrs()` sets `0` if driver is not powered by VMOT

rubienr, 2020-12-07:
### Bug Description

When TMC2130 stepper driver is not powered by VMOT (i.e. PSU_CONTROL is enabled and PSU is off), 
`TMCMarlin::set_pwm_thrs(const uint32_t)` will call ` TMC::TPWMTHRS(0)` since `this->microsteps()` returns the readout of the unpowered driver's internal processor, not a shadowed register ([see tmc_util.h:116](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/tmc_util.h#L116) and all subsequent implementations of `TMCMarlin::set_pwm_thrs(const uint32_t thrs)`)

The rough call tree:
```
TMCMarlin::set_pwm_thrs(const uint32_t)
  -> uint16_t TMCStepper::microsteps() 
    -> uint8_t TMC2130Stepper::mres() { CHOPCONF_t r{0}; r.sr = CHOPCONF(); return r.mres; }
      -> uint32_t TMC2130Stepper::CHOPCONF() { return read(CHOPCONF_register.address); } // attempts to read from the controller
  -> TMC::TPWMTHRS(_tmc_thrs(this->microsteps() /* == 0 if unpowered */, thrs, planner.settings.axis_steps_per_mm[AXIS_ID]));
    -> constexpr uint16_t _tmc_thrs(const uint16_t msteps /* == 0 */, const uint32_t thrs, const uint32_t spmm)
         {return 12650000UL*msteps /* msteps == 0 */ /(256*thrs*spmm);}
```
### Configuration Files

Enable `#define PSU_CONTROL` and at least one TMC driver: `#define *_DRIVER_TYPE  TMC2130`.

### Steps to Reproduce

1. start Marlin (assuming a non `0` hybrid threshold value is stored in flash) 
2. make sure PSU is off
3. trigger `TMCMarlin::set_pwm_thrs(const uint32_t)` for example by `M913 X100`
4. enable PSU `M80`
5. `M913` will report `0` 

**Expected behaviour:**
1.-3.
4. enable PSU `M80` (expect that drives are re-initialized, for example `TMCMarlin::set_pwm_thrs(const uint32_t)` is called ...)
5. `M913` will report the hybrid threshold stored in flash
 
**Actual behaviour:**
1.-4.
5. `M913` will report `0`

### Probably Related

https://github.com/MarlinFirmware/Marlin/pull/19364 (PR: Use stored values for TMC StealthChop)
https://github.com/MarlinFirmware/Marlin/issues/19313 (Issue: [BUG] M502/M500 when TMC drivers do not have VMOT results in incorrect values being stored in EEPROM)

ManuelMcLure, 2020-12-07:
I've been discussing this on Discord with @rubienr and agree that this is a bug, This was something that slipped through the cracks when I looked at #19364 - the hybrid threshold is stored in a shadow register in TMCStepper, but the micro steps are not and those are used to initialize the shadow register hence the bug.

ManuelMcLure, 2020-12-07:
The micro step setting is indirectly shadowed on the TMCStepper side as part of `CHOPCONF` for re-initializing the drivers, but reading it always goes to the driver, which is the design pattern for TMCStepper. On the Marlin side we don't keep a shadow version of the micro step setting, probably because it can only be set in `Configuration_adv.h` (no gcode or EEPROM storage) so it was not considered worth the RAM.
Unfortunately at the point where we update the hybrid threshold we don't have access to the original value for micro steps that was passed in to `tmc_init()` so it looks like to fix this we will have to add extra storage on the Marlin side.

rubienr, 2020-12-07:
Tested with bugfix-2.0.x as of [9b82a5c](https://github.com/rubienr/MarlinFirmware/tree/9b82a5cd767fc0fa864174f9333b8423afd7f3f4) and observed the same behaviour as described above.

### Configuration Files

[Configuration.h](https://github.com/rubienr/MarlinFirmware/blob/9b82a5cd767fc0fa864174f9333b8423afd7f3f4/private/config/minimal-config/Configuration.h)
[Configuration_adv.h](https://github.com/rubienr/MarlinFirmware/blob/9b82a5cd767fc0fa864174f9333b8423afd7f3f4/private/config/minimal-config/Configuration_adv.h)

ManuelMcLure, 2020-12-07:
I'm pretty sure this is not specific to the 2130 drivers and will occur with all TMC drivers.

ClockeNessMnstr, 2020-12-27:
Not sure if this is part of the same thing or maybe expected behavior for some reason but I was also getting '0' set by "M913 E5" for my TMC2209 on my extruder which had microstepping disabled / = 0.
it worked when I set microsteps to 4. I was able to set M913 E5.
Other motors worked set at 16. 

Seems to set '0' if microsteps is set to 0 for the driver. I don't quite understand the firmware enough to know if h

rubienr, 2021-01-29:
If needed I can offer my printer for remote debugging/flashing.

closed 2021-03-17 by github-actions[bot]