MarlinFirmware/Marlin #20399: [FR] Filament runout / Advanced_Pause_Feature without LCD

italocjs, 2020-12-07:
Same as this FR #16848  

I'm using repetier server with a screen, and no onboard screen,.  when i try to setup the filament sensor it requires advanced pause, which in turn requires a LCD which i dont have connected to the board.

Any chance to make this independent? 

Error: 
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:839:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

ETE-Design, 2020-12-07:
Have you tryed to enable EMERGENCY_PARSER as it suggest when you compile?

italocjs, 2020-12-08:
Enabling emergency parser, Load and unload gcodes, i get the following (seems unrelated but if i comment those features the errors go away)

Marlin\src\feature\pause.cpp: In function 'bool ensure_safe_temperature(bool, PauseMode)':
Marlin\src\feature\pause.cpp:136:51: error: 'class Temperature' has no member named 'extrude_min_temp'

italocjs, 2020-12-08:
i believe something in the code is only enabled if a lcd is present

ETE-Design, 2020-12-08:
@italocjs It is possible to enable without LCD, it is a known bug but if you set EXTRUDE_MINTEMP to 0 you can overcome it, but of cause on your own risk :-)

italocjs, 2020-12-08:
Thanks, i'll try it tomorrow first thing!

ETE-Design, 2020-12-18:
@italocjs Did it work? And if so then pls. Close this issue...

italocjs, 2020-12-18:
Sorry, forgot to close it.
It did work!

closed 2020-12-18 by italocjs