MarlinFirmware/Marlin #20402: [BUG] Simple  the Latest release  Platform  2.0.7.2	2.0.x.zip  Config H states For a SCARA printer start with the configuration files in // config/examples/SCARA and customize for your machine.  Yet the config folder in the downloaded zip file only contains a readme file.   But the config folder on gethub has the example folder

phearl3ss1, 2020-12-07:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

phearl3ss1, 2020-12-07:
However when compiling with config and adv. config  EXAMPLES_DIR "SCARA/MP_SCARA" replaced 
 there so many compiling errors I wouldn't know where to begin and I doubt the example has been updated to Marlin 2.0

thisiskeithb, 2020-12-07:
Can you shorten the title and put that in the body of the issue template? Please follow the issue template next time.

I've created PR https://github.com/MarlinFirmware/Marlin/pull/20403 to adress the inline documentation.

> However when compiling with config and adv. config EXAMPLES_DIR "SCARA/MP_SCARA" replaced
there so many compiling errors I wouldn't know where to begin and I doubt the example has been updated to Marlin 2.0

I just tried building both default SCARA configs from the [MarlinFirmware/Configurations/tree/import-2.0.x](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config) branch with the latest Marlin `bugfix-2.0.x` and both built correctly.

phearl3ss1, 2020-12-07:
I will try it.   That is different from the one on line that I assumed should have been in the zip file. 
  The on line files that did not compile are in this folder Configurations/config/examples/SCARA/   not "import-2.0x" 
  Thankyou

thisiskeithb, 2020-12-07:
> I will try it. That is different from the one on line that I assumed should have been in the zip file.
> The on line files that did not compile are in this folder Configurations/config/examples/SCARA/ not "import-2.0x"
> Thankyou

You need to match the version of Marlin you're compiling with the version of configs. So if you were trying the default (which is the import-2.0.x branch) with an older version of Marlin, it will likely fail to compile.

sjasonsmith, 2020-12-08:
It is probably safe to assume you are going to run into issues configuring a Scara printer. Please be prepared to investigate issues and possibly contribute fixes in the process.

thisiskeithb, 2020-12-09:
Closing since documentation has been updated (and configs build as-is without errors).

closed 2020-12-09 by thisiskeithb