MarlinFirmware/Marlin #20406: [BUG] Re-bump don't work with sensorless on z axis

alexandremstf, 2020-12-08:
### Bug Description

Re-bump don't work on z axis with sensorless. On x and y axis works properly.

**Using M111 S237:**

```
Home 1 Fast:
do_homing_move  X-18.00 Y-10.00 Z320.00
...(Z, 480.00, [20.00])
<<< do_homing_move  X-18.00 Y-10.00 Z320.00
Move Away:
>>> do_homing_move  X-18.00 Y-10.00 Z320.00
...(Z, -20.00, [20.00])
<<< do_homing_move  X-18.00 Y-10.00 Z320.00
Home 2 Slow:
>>> do_homing_move  X-18.00 Y-10.00 Z320.00
...(Z, 40.00, 20.00)
```

**"Home 1 Fast" - work properly
"Move Away" - don't work properly
"Home 2 Slow" - work properly**

For some reason **Move Away** is being ignored.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5655970/Marlin.zip)

### Steps to Reproduce

1. Set sensorless homing
2. Dual z axis
3. Dual z endstops
4. Home Dir 1 (Max)

**Expected behavior:**

Re-bump on z axis

alexandremstf, 2020-12-08:
Reporting that is not a bug, i made a mistake and defined USE_ZMIN_PLUG and USE_ZMAX_PLUG simultaneously, because of this the re-bump didn't work on Z axis.

sjasonsmith, 2020-12-08:
Re-opening. You had a slight misconfig, but I don't think USE_ZMAX_PLUG should not interfere with homing to -Z.

Issues involving this have appeared several times. I would like to see if we can find a way to improve the behavior, or inform users of the problem.

alexandremstf, 2020-12-08:
Yes, you're right, in my case when define USE_ZMIN_PLUG, make my homing to z+ problematic. For some reason that i couldn't find the re-bump does not occurs.

sjasonsmith, 2020-12-09:
My thought is that with both defined it seems that both endstops are triggered at the same time, preventing movement to back away from the endstop. I do not believe this is unique sensorless homing.

When this occurs we need to fail in some way, perhaps even calling kill(). Proceeding with movement after a failed backoff has the potential to damage hardware.

Maybe we can detect issues at build time, but I believe having both min/max endstops is a possibly valid configuration. We may instead need a runtime check to verify that both endstops are not triggered at the same time.

alexandremstf, 2020-12-09:
It is realy difficult to detect an issue, because i can use USE_ZMIN_PLUG and USE_ZMAX_PLUG in other words min/max endstops at the same  time like in dual z mode.

alexandremstf, 2020-12-09:
I think the problem is what you said, endstops triggered at the same time. Maybe perform a different treatment on the endstop if dual or multi endstops are been used for the same axis.

closed 2021-01-16 by github-actions[bot]