MarlinFirmware/Marlin #20407: CRASH AT STARTUP WITH MMU2

srastoul, 2020-12-08:
### Bug Description

When I switch on my printer it crashes during startup after the logo appears. If I do a reset it starts and everything works. See video attached with configuration files

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5661496/Marlin.zip)

#### Additional Information

BORD MKS SGEN L V2

GMagician, 2020-12-13:
It seems like a watchdog trigger. Maybe Marlin is waiting for a "never sent" answer. Once MMU is already switched on then the anaswer comes quickly.
To confirm, if you have possibility, is to turn on mmu and once started turn on your printer

srastoul, 2020-12-13:
It is difficult to put the power on the MMU before the printer because the 5V is supplied by the printer board. I tried by deactivating the watchdog in configuration_adv and it's the same problem

srastoul, 2020-12-13:
I found a regulator to generate 5V and did the test. If i feed the mmu before the printer all starts up correctly.

srastoul, 2020-12-13:
Before, my mmu was installed on a printer with marlin 2.0.1 and an MKS GEN L V1 card and there was no problem.

GMagician, 2020-12-13:
Sure your 5V power has enough strenght ? Maybe once both are switched on there is a drop in voltage

srastoul, 2020-12-13:
For me the problem is not on the 5 V I measured it and it does not lower. I have just done a test again if the mmu is powered by my external regulator and I turn on everything simultaneously there is the problem.

GMagician, 2020-12-13:
The only thing that can reset your printer, AFAIK, is the WD, but you disabled it.
If we remove the power issue then the only thing that may trigger this may be the startup sequence.
I don't have such HW so it's a little bit difficult to check what may happens. I'll give a look at the code

GMagician, 2020-12-13:
Try to enable MMU2_DEBUG and the check what you get on host console

srastoul, 2020-12-14:
Here is a screenshot of the console at startup:

Connecting...
Connecting...
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 T0 D1.75
echo:  M200 T1 D1.75
echo:  M200 T2 D1.75
echo:  M200 T3 D1.75
echo:  M200 T4 D1.75
echo:  M200 S0
echo:; Steps per unit:
echo: M92 X100.00 Y100.00 Z400.00 E830.00
echo:; Maximum feedrates (units/s):
echo:  M203 X250.00 Y250.00 Z35.00 E80.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z500.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T1500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z10.00
echo:; Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:  M145 S2 H240 B80 F0
echo:; PID settings:
echo:  M301 P14.40 I0.70 D73.50
echo:  M304 P41.50 I2.33 D493.10
; Controller Fan
echo:  M710 S120 I0 A1 D60 ; (47% 0%)
echo:; Power-Loss Recovery:
echo:  M413 S1
echo:; Z-Probe Offset (mm):
echo:  M851 X26.00 Y4.00 Z-0.90
echo:; Stepper driver current:
echo:  M906 X700 Y700 Z700
echo:  M906 I1 Z700
echo:  M906 T0 E600
echo:; StallGuard threshold:
echo:  M914 X1 Y1
echo:; Driver stepping mode:
echo:; Linear Advance:
echo:  M900 T0 K0.00
echo:  M900 T1 K0.00
echo:  M900 T2 K0.00
echo:  M900 T3 K0.00
echo:  M900 T4 K0.00
echo:; Filament load/unload lengths:
echo:  M603 T0 L60.00 U120.00
echo:  M603 T1 L60.00 U120.00
echo:  M603 T2 L60.00 U120.00
echo:  M603 T3 L60.00 U120.00
echo:  M603 T4 L60.00 U120.00
echo:; Tool-changing:
echo: Z2.00
echo:; Filament runout sensor:
echo:  M412 S1
Testing X connection... OK
Testing Y connection... OK
Testing Z conn
[ERROR] Can't write to printer (disconnected?) (SerialException): Write timeout
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed

srastoul, 2020-12-14:
And after reset printer:

Connecting...
Connecting...
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 T0 D1.75
echo:  M200 T1 D1.75
echo:  M200 T2 D1.75
echo:  M200 T3 D1.75
echo:  M200 T4 D1.75
echo:  M200 S0
echo:; Steps per unit:
echo: M92 X100.00 Y100.00 Z400.00 E830.00
echo:; Maximum feedrates (units/s):
echo:  M203 X250.00 Y250.00 Z35.00 E80.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z500.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T1500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z10.00
echo:; Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:  M145 S2 H240 B80 F0
echo:; PID settings:
echo:  M301 P14.40 I0.70 D73.50
echo:  M304 P41.50 I2.33 D493.10
; Controller Fan
echo:  M710 S120 I0 A1 D60 ; (47% 0%)
echo:; Power-Loss Recovery:
echo:  M413 S1
echo:; Z-Probe Offset (mm):
echo:  M851 X26.00 Y4.00 Z-0.90
echo:; Stepper driver current:
echo:  M906 X700 Y700 Z700
echo:  M906 I1 Z700
echo:  M906 T0 E600
echo:; StallGuard threshold:
echo:  M914 X1 Y1
echo:; Driver stepping mode:
echo:; Linear Advance:
echo:  M900 T0 K0.00
echo:  M900 T1 K0.00
echo:  M900 T2 K0.00
echo:  M900 T3 K0.00
echo:  M900 T4 K0.00
echo:; Filament load/unload lengths:
echo:  M603 T0 L60.00 U120.00
echo:  M603 T1 L60.00 U120.00
echo:  M603 T2 L60.00 U120.00
echo:  M603 T3 L60.00 U120.00
echo:  M603 T4 L60.00 U120.00
echo:; Tool-changing:
echo: Z2.00
echo:; Filament runout sensor:
echo:  M412 S1
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
MMU <= reset
echo:SD card ok
Printer is now online.
MMU => 'start'
MMU <= 'S1'
MMU => 106
MMU <= 'S2'
MMU => 372
MMU <= 'P0'
MMU => 0
MMU - ENABLED

srastoul, 2020-12-14:
And if i unplug RX and TX MMU connection:

Connecting...
Connecting...
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 T0 D1.75
echo:  M200 T1 D1.75
echo:  M200 T2 D1.75
echo:  M200 T3 D1.75
echo:  M200 T4 D1.75
echo:  M200 S0
echo:; Steps per unit:
echo: M92 X100.00 Y100.00 Z400.00 E830.00
echo:; Maximum feedrates (units/s):
echo:  M203 X250.00 Y250.00 Z35.00 E80.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z500.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1500.00 R1500.00 T1500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z10.00
echo:; Material heatup parameters:
echo:  M145 S0 H200 B60 F0
echo:  M145 S1 H240 B110 F0
echo:  M145 S2 H240 B80 F0
echo:; PID settings:
echo:  M301 P14.40 I0.70 D73.50
echo:  M304 P41.50 I2.33 D493.10
; Controller Fan
echo:  M710 S120 I0 A1 D60 ; (47% 0%)
echo:; Power-Loss Recovery:
echo:  M413 S1
echo:; Z-Probe Offset (mm):
echo:  M851 X26.00 Y4.00 Z-0.90
echo:; Stepper driver current:
echo:  M906 X700 Y700 Z700
echo:  M906 I1 Z700
echo:  M906 T0 E600
echo:; StallGuard threshold:
echo:  M914 X1 Y1
echo:; Driver stepping mode:
echo:; Linear Advance:
echo:  M900 T0 K0.00
echo:  M900 T1 K0.00
echo:  M900 T2 K0.00
echo:  M900 T3 K0.00
echo:  M900 T4 K0.00
echo:; Filament load/unload lengths:
echo:  M603 T0 L60.00 U120.00
echo:  M603 T1 L60.00 U120.00
echo:  M603 T2 L60.00 U120.00
echo:  M603 T3 L60.00 U120.00
echo:  M603 T4 L60.00 U120.00
echo:; Tool-changing:
echo: Z2.00
echo:; Filament runout sensor:
echo:  M412 S1
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
MMU <= reset
echo:SD card ok
Printer is now online.

GMagician, 2020-12-14:
It's really seems that something happens on tmc serial communication during startup. I can't figure out what

closed 2021-01-21 by github-actions[bot]