MarlinFirmware/Marlin #20409: [BUG] SKR1.4T microstepping not working

tlloydjonesprog, 2020-12-09:
1. Flash with FW
1b. optional Init/Save
2. Start a print
3. microstepping does not work

**Expected behavior:**

Expected it to step as needed to adjust Z offset and make the hot-end lower or father than expected

**Actual behavior:**
No movement even if offset if changed and saved until print is stopped and restarted

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5663014/Configuration_adv.zip)

thisiskeithb, 2020-12-09:
Do you mean babystepping? If so, try enabling `INTEGRATED_BABYSTEPPING`.

tlloydjonesprog, 2020-12-09:
> Do you mean babystepping? If so, try enabling `INTEGRATED_BABYSTEPPING`.

I did; apologies! Tried that once already but no luck; I'll try again though.

ellensp, 2020-12-09:
When you say flash firmware are you just hitting upload, as these boards require you to reboot after uploading to actually flash new firmware.

tlloydjonesprog, 2020-12-09:
Yup, full reboot. Process has been compile -> Rename bin as needed -> put into printer when it's off -> turn on -> Wait for blue screen to end finishing fw update -> home -> start print -> edit offset

Still no luck with probe offset

ChaChaBwoy3D, 2020-12-10:
> Yup, full reboot. Process has been compile -> Rename bin as needed -> put into printer when it's off -> turn on -> Wait for blue screen to end finishing fw update -> home -> start print -> edit offset
> 
> Still no luck with probe offset

I'm using same board and I can babystep my nozzle just fine recheck your configuration.

tlloydjonesprog, 2020-12-10:
> I'm using same board and I can babystep my nozzle just fine recheck your configuration.

With latest branch/bugfix? If so, can you upload your config. I'd love to know what I did wrong

ChaChaBwoy3D, 2020-12-10:
Sure thing
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5669931/Configuration.zip)

tlloydjonesprog, 2020-12-10:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5670735/Marlin.zip)
Compared and updated mine but still no luck. Also tends to be slow on the LCD during prints. Z offset doesn't change unless the print is restarted

tlloydjonesprog, 2020-12-16:
Anything I can do to help fix?

rhapsodyv, 2020-12-16:
I have marlin installed in a skr 1.4 turbo + btt tft 3.0. My marlin is for two weeks ago.
And I can do babystep during the print.
As I do a lot of testing, I frequently forget to do z offset before printing. So I use a lot this function: babystep while printing.

How are you doing babystep? By the lcd? By gcode commands? By the BTT TFT?

tlloydjonesprog, 2020-12-17:
LCD - tried gcode too but no luck

rhapsodyv, 2020-12-17:
Can you try this?
```cpp
#define BABYSTEP_MILLIMETER_UNITS
#define BABYSTEP_MULTIPLICATOR_Z 0.1
```

tlloydjonesprog, 2020-12-19:
> Can you try this?
> 
> ```c++
> #define BABYSTEP_MILLIMETER_UNITS
> #define BABYSTEP_MULTIPLICATOR_Z 0.1
> ```

That worked - either that or disabling menu for PID autotune on LCD :) Many thanks.

closed 2020-12-19 by tlloydjonesprog