MarlinFirmware/Marlin #20411: [BUG] No longer possible to have a probe which triggers above the nozzle

c3D-Dan, 2020-12-09:
Using latest bugfix 2.x

From Sanity check 
static_assert(sanity_nozzle_to_probe_offset.z <= 0.25,
                "Are you sure your Probe triggers above the nozzle? Set a negative Z value in the NOZZLE_TO_PROBE_OFFSET.");

Yes I am sure my probe triggers above the nozzle.  This would be true with any configuration where the nozzle moves out of the way to expose a probe which is above the nozzle height.  Forcing users into specific probe configurations seems counter productive. Im all up for a warning or notice in this regard, but not a compilation error. 

For the moment, I have commented the offending line in  SanityCheck.h

and relevant settings from configuration.h
#define NOZZLE_TO_PROBE_OFFSET { 0, 0, 7.0 } //My nozzle is 7mm lower than my probe.

sjasonsmith, 2020-12-09:
There is a pull request open related to this: #20344

Could you describe what your hardware is like, to help educate us on examples where this may be used? Links or pictures showing the setup would be helpful as well.

In the future, please be sure to adhere to the template provided when filing a bug report, including providing your configuration files. That consistency makes it much easier for people who are triaging and debugging issues.

sjasonsmith, 2020-12-09:
@c3D-Dan please post your configuration files. We can use them to help us create a test case to avoid breaking configurations such as yours in the future.

sjasonsmith, 2020-12-09:
Please test again with the latest bugfix, and confirm whether your problem is solved.

c3D-Dan, 2020-12-09:
Sure thing.  I will tomorrow morning.  On my phone now.

In reply to your question about what hardware: 
http://Cultivate3D.com  specifically look at the sections on “Medusa”.

sjasonsmith, 2020-12-19:
I am closing that as presumably this has been fixed. Please reply back if that is not true after your testing.

closed 2020-12-19 by sjasonsmith