MarlinFirmware/Marlin #20413: [BUG] J GAURORA A5S Not save eeprom settings on sdcard. 2.0.7.3

jbaiao, 2020-12-09:
### Bug Description

JGAURORA A5S. 
Not save eeprom settings on sdcard.
On boot say´s "Save Config" on lcd screen. 

### Configuration Files

Use default settings (donwload from github marlin) for that machine for marlin 2.0.7.2

### Steps to Reproduce

1. Put firmware on sd format as FAT32
2. Reboot machine and install firmware
3. After reboot apears "Save configuration"
4. Change something and "Save to memory"
5. Reboot
6. Nothing change

jbaiao, 2020-12-10:
Already change sd card, remade firmware, and its still the problem.
Every time the printer is restarted, the message "Store settings" appears on the LCD and any value that has changed is reset as if the command M502 and M500 had been sent.
I also tried to compile with ABL but after starting to print the values of G29 are not used, it seems that they are inverted ....
Please ... help!
:)

sjasonsmith, 2020-12-22:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Please be sure to test with the bugfix-2.0.x example as well:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/JGAurora/A5S

sjasonsmith, 2020-12-31:
Closing. Original submitter is unresponsive. I know this is working on other STM32F1 boards in `bugfix-2.0.x`, because I fixed an issue impacting Creality boards. I don't have this exact board to test with, but I don't believe anything should make it unique.

@jbaiao if you re-test with `bugfix-2.0.x` and continue to have issues, this may be re-opened.

closed 2020-12-31 by sjasonsmith