MarlinFirmware/Marlin #20415: Anet A8 plus на btt skr 1.4 и TMC 2208 v2.0

y3kov, 2020-12-09:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.
 Compiling .pio\build\LPC1768\src\src\module\motion.cpp.o

In file included from Marlin\src\module\endstops.cpp:36:

Marlin\src\module\../HAL/LPC1768/endstop_interrupts.h:55:8: error: #error "X_MIN_PIN is not INTERRUPT-capable."

   55 | #error "X_MIN_PIN is not INTERRUPT-capable."

      | ^~~~~

Marlin\src\module\../HAL/LPC1768/endstop_interrupts.h:67:8: error: #error "Y_MIN_PIN is not INTERRUPT-capable."

   67 | #error "Y_MIN_PIN is not INTERRUPT-capable."

      | ^~~~~

Marlin\src\module\../HAL/LPC1768/endstop_interrupts.h:79:8: error: #error "Z_MIN_PIN is not INTERRUPT-capable."

   79 | #error "Z_MIN_PIN is not INTERRUPT-capable."

      | ^~~~~

Compiling .pio\build\LPC1768\src\src\module\planner.cpp.o

Compiling .pio\build\LPC1768\src\src\module\servo.cpp.o

*** [.pio\build\LPC1768\src\src\module\endstops.cpp.o] Error 1

========================================================== [FAILED] Took 250.57 seconds

sjasonsmith, 2020-12-09:
Disable `ENDSTOP_INTERRUPTS`.

We should change this error message to provide that instruction.

ellensp, 2020-12-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sjasonsmith, 2020-12-09:
@ellensp I am going to re-open this as a reminder to update the error message. That will allow future users who encounter this to solve their issue without seeking help.

thisiskeithb, 2020-12-10:
Added here: https://github.com/MarlinFirmware/Marlin/pull/20419.

closed 2020-12-10 by thisiskeithb