MarlinFirmware/Marlin #20416: [BUG] (Probe Offset Wizard currently broken due to PROBE_OFFSET_WIZARD_START_Z)

LearnIIBurn, 2020-12-09:
### Bug Description

My first "bug" report!

This may be some in-progress changes to the probe offset wizard, but currently Marlin will not build with the wizard enabled because there is no declaration possible of function `PROBE_OFFSET_WIZARD_START_Z`. It requires a negative value to commence. Also I know I butchered common programming language, pardon on that.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5668110/Configs.zip)

### Steps to Reproduce

1. Enable Probe Offset Wizard in Configuration_adv.h
2. Uncomment (or don't) `#define PROBE_OFFSET_WIZARD_START_Z -4`
3. Build Marlin

**Expected behavior:**

Marlin builds successfully

**Actual behavior:**

Fails build with `#error directive: "PROBE_OFFSET_WIZARD requires a PROBE_OFFSET_START with a negative value."`

#### Additional Information

Changing `PROBE_OFFSET_WIZARD_START_Z` back to `PROBE_OFFSET_START` and declaring a negative value seems to fix it in the interim.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

swissnorp, 2020-12-10:
In latest bugfix-2.0.x there is no such check in SanityCheck.h:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/SanityCheck.h

Update your Fork and test again! Its not enough if you just take the new configuration files. 😉

LearnIIBurn, 2020-12-11:
Sorry about that I see my mistake! Thank you for the guidance!

closed 2020-12-11 by LearnIIBurn

petitxef, 2020-12-14:
@LearnIIBurn could you share how to fix it? I've the same error and I just downloaded the last bugfix and didn't work. Thanks