MarlinFirmware/Marlin #20417: Serial Communication has Problems because ubl.cpp is missing a newline.

unixhelden, 2020-12-09:
After compiling and uploading todays bugfix release i noticed i had massive issues with octoprint.

After asking for help in Octoprint Forum @foosel noticed there is a problem with Marlin where it does not set the Newline for a message correctly. 

diff --git a/Marlin/src/feature/bedlevel/ubl/ubl.cpp b/Marlin/src/feature/bedlevel/ubl/ubl.cpp
index 087fdf42b2..82ce79e025 100644
--- a/Marlin/src/feature/bedlevel/ubl/ubl.cpp
+++ b/Marlin/src/feature/bedlevel/ubl/ubl.cpp
@@ -43,7 +43,7 @@
   #include "math.h"

   void unified_bed_leveling::echo_name() {
-    SERIAL_ECHOPGM("Unified Bed Leveling");
+    SERIAL_ECHOPGM("Unified Bed Leveling\n");
   }

   void unified_bed_leveling::report_current_mesh() {

This Patch solves the Problem. 

Thx in advance :)

ellensp, 2020-12-10:
If this needs a new line, its using the wrong macro
should use SERIAL_ECHOLNPGM, this add the \n automatically.

ellensp, 2020-12-10:
But I believe you are in error.

echo_name() is defined as follows

```
  void unified_bed_leveling::echo_name() {
    SERIAL_ECHOPGM("Unified Bed Leveling");
  }
```

It is only used in two places
```
      if (!sanity_check()) {
        echo_name();
        SERIAL_ECHOLNPGM(" sanity checks passed.");
      }
```
This is clearly printing "Unified Bed Leveling sanity checks passed.\n" as one line, not two

The second place is in the following
```
  void unified_bed_leveling::report_state() {
    echo_name();
    SERIAL_ECHO_TERNARY(planner.leveling_active, " System v" UBL_VERSION " ", "", "in", "active\n");
    serial_delay(50);
  }
```
UBL_VERSION is "1.01"
This is printing "Unified Bed Leveling System v1.01 "
then ether "active\n" or "inactive\n"
Once again a complete line.

No missing new lines here.

ellensp, 2020-12-10:
more details here https://community.octoprint.org/t/serial-communication-with-printer-not-reliable/27339

ellensp, 2020-12-10:
The following seem to be in error, when DEBUG_ECHOLNPGM is disabled

 ```
            ubl.echo_name();
              DEBUG_ECHOLNPGM(" initialized.\n"); ```

ellensp, 2020-12-10:
A better answer is find the following in Marlin/src/module/settings.cpp
`//#define DEBUG_OUT 1`
change to 
`#define DEBUG_OUT ENABLED(DEBUG_LEVELING_FEATURE)`
then find 
```
            #if ENABLED(EEPROM_CHITCHAT)
              ubl.echo_name();
              DEBUG_ECHOLNPGM(" initialized.\n");
            #endif
```
and change it to
```
            #if BOTH(EEPROM_CHITCHAT, DEBUG_LEVELING_FEATURE)
              ubl.echo_name();
              DEBUG_ECHOLNPGM(" initialized.\n");
            #endif
```
then find 
```
            #if ENABLED(EEPROM_CHITCHAT)
              DEBUG_ECHOPGM("?Can't enable ");
              ubl.echo_name();
```
and change it to
```
            #if BOTH(EEPROM_CHITCHAT, DEBUG_LEVELING_FEATURE)
              DEBUG_ECHOPGM("?Can't enable ");
              ubl.echo_name();
```

ellensp, 2020-12-10:
PR created

ellensp, 2020-12-10:
This has now been merged so is in bugfix.

closed 2020-12-10 by ellensp