MarlinFirmware/Marlin #20418: [BUG] LPC176x (SKR 1.4 Turbo) boards and MAX31855 not working on E1 empty Driver Socket in Software SPI mode

GadgetAngel, 2020-12-09:
### Bug Description

Since the SKR V1.3 board and the SKR V1.4 and SKR V1.4 Turbo boards do not have a lot of extra free pins, I want to use the free pins on the empty E1 Driver socket to run a K-Type Thermocouple temperature sensor. I do this on the SKR PRO and GTR boards without an issue.

I have an SKR V1.4 Turbo board I am trying to hook up a K-Type thermocouple with a MAX31855 board from Adafruit.  When I use the empty E1 driver socket pins (E1_DIR_PIN , E1_STEP_PIN and E1_PDN_PIN) all I get is MIN TEMP error.  So I double check my work and I still get MIN TEMP error.  I went and bought myself a USB logic analyzer from Saleae (PRO 8 channel).  I will be supplying the capture SPI bus data with this bug report.  

My SKR V1.4 Turbo board is using 3 UART drivers TMC2226 in X, Y, and Z.  The E0 Driver is in SPI mode and is a TMC2130.  I ensure I comment out the E1 pins defined in the pins_BTT_SKR_V1_4.h file and in pins_BTT_SKR_common.h file.  I comment out the CS_PIN for all drivers not using SPI mode.  I comment out ?_SERIAL_TX_PIN and ?_SERIAL_RX_PIN for all drivers not in UART mode:
pins_SKR_V1_4.h file:
```
//
// Using UART mode on X, Y and Z BUT using SPI mode on E0, E1 NOT being used
//

#define X_STEP_PIN                         P2_02
#define X_DIR_PIN                          P2_06
#define X_ENABLE_PIN                       P2_01
//#ifndef X_CS_PIN
  //#define X_CS_PIN                         P1_10
//#endif

#define Y_STEP_PIN                         P0_19
#define Y_DIR_PIN                          P0_20
#define Y_ENABLE_PIN                       P2_08
//#ifndef Y_CS_PIN
  //#define Y_CS_PIN                         P1_09
//#endif

#define Z_STEP_PIN                         P0_22
#define Z_DIR_PIN                          P2_11
#define Z_ENABLE_PIN                       P0_21
//#ifndef Z_CS_PIN
  //#define Z_CS_PIN                         P1_08
//#endif

#define E0_STEP_PIN                        P2_13
#define E0_DIR_PIN                         P0_11
#define E0_ENABLE_PIN                      P2_12
#ifndef E0_CS_PIN
  #define E0_CS_PIN                        P1_04
#endif

/*
#define E1_STEP_PIN                        P1_15
#define E1_DIR_PIN                         P1_14
#define E1_ENABLE_PIN                      P1_16
//#ifndef E1_CS_PIN
  //#define E1_CS_PIN                        P1_01
//#endif
*/

//#define TEMP_1_PIN                      P0_23_A0  // A0 (T0) - (67) - TEMP_1_PIN
#define TEMP_BED_PIN                    P0_25_A2  // A2 (T2) - (69) - TEMP_BED_PIN

//
// Software SPI pins for TMC2130 stepper drivers
//
#if ENABLED(TMC_USE_SW_SPI)
  #ifndef TMC_SW_MOSI
    #define TMC_SW_MOSI                    P1_17
  #endif
  #ifndef TMC_SW_MISO
    #define TMC_SW_MISO                    P0_05
  #endif
  #ifndef TMC_SW_SCK
    #define TMC_SW_SCK                     P0_04
  #endif
#endif

#if HAS_TMC_UART
  /**
[... 222 lines cut ...]
I expect my printer to use the temperature readings from the thermocouple and NOT to get and MIN TEMP error on boot up.

**Actual behavior:**

I get a MIN TEMP error! on bootup of the SKR V1.4 Turbo board.

#### Additional Information

Here are the SPI capture files I was describing above.  Did you all know that the **first clock pulse** and the **second clock pulse** produced by the LPC176x board **is less than 100 ns apart**.  The MAX31855 datasheet and other SPI data sheets all say that a minimum of 100 ns on the negative clock pulse width is needed. You will see this in the SPI capture data sets.

You can download the Saleae Software (2.3.15) from here and use it to open up these files to look at them for yourself: https://ideas.saleae.com/f/changelog/

Documents on how to use the Saleae Software can be found here: https://support.saleae.com/getting-started

Test number 1 SPI capture: 
[Test number 1 SPI capture data set.zip](https://github.com/MarlinFirmware/Marlin/files/5669116/P1_01.acting.as.SCK.line.with.P1_24.as.CS.line.and.P0_03.as.DO_PIN.appears.to.be.OK.zip)

Test number 2 SPI capture: THIS IS where you see a problem (ONLY ONE SAMPLE is allowed through!): 
[Test number 2 SPI capture data set.zip](https://github.com/MarlinFirmware/Marlin/files/5669128/P1_01.acting.as.D0_PIN.with.P1_24.as.CS.and.P0_02.as.SCK.ONE.reading.then.0.zip)

Test number 3 SPI capture: 
[Test number 3 SPI capture data set.zip](https://github.com/MarlinFirmware/Marlin/files/5669133/P1_15.acting.as.SCK_PIN.with.P1_24.as.CS.and.P0_03.as.DO_PIN.everything.appears.OK.zip)

Test number 4 SPI capture: THIS IS where you see a problem (ONLY ONE SAMPLE is allowed through!)  Now that this happened a second time with a different pin the problem is repeatable.  
[Test number 4 SPI capture data set.zip](https://github.com/MarlinFirmware/Marlin/files/5669146/P1_15.acting.as.DO_PIN.with.P1_24.acting.as.CS.and.P0_02.acting.as.SCK_PIN.ONLY.one.sample.then.zero.values.zip)

rhapsodyv, 2020-12-10:
It's not the same as #20359, but on different pins? Looks like the problem is SW with MAX sensors on any pin set...

GadgetAngel, 2020-12-10:
@rhapsodyv BUG report #20359 is about the MAX6675 dropping temperature readings down to 0 for no known reason.  

This BUG report is the inability to use the empty E1 driver socket pins to run a MAX31855 board. 

The next report you will see is #20429 which is the inability to use Hardware SPI with RX0 pin on the TFT connector.  

It feels like the LPC176x boards are running on the edge of getting data with SPI.  If you look at the SPI captures I sent with this bug report you will see that the clock pulses are not the recommended 100ns pulse width.  

I will be submitting a PR that will fix this bug report and #20429 bug report.   They both are fixed by using the Adafruit MAX31855 library for the LPC1768 board.  The Adafruit MAX31855 library spaces the clock pulses to 1ms instead of less than 100ns.  I am testing my PR to ensure that it does not cause more issues to come up.

rhapsodyv, 2020-12-10:
You reported 4 issues about MAX sensors (31855 or 6675), all on LPC.... I really think that they share the same cause.

But as I soon I receive mine, I will take a deep look. I think they will arrive only after holidays.

GadgetAngel, 2021-01-12:
> This issue has had no activity in the last 30 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 7 days.

This issue is still under review because the pull request that fixes this issue is still going through the review process.

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by GadgetAngel