MarlinFirmware/Marlin #20424: No extruders (motors) but hotends (PID control)

Daniele-Dondi, 2020-12-10:
### Description

Hi, I'm quite new here. I'm trying to use Marlin on my robot for chemical reactions. Actually the robot has 6 axes for driving syringes plus several servos for valves. I wish to implement the possibility to heat (by using PID) but the only way I found is to enable one extruder (that I don't need and also I haven't enough pins to use on my board).
So, I'm asking if it is possible to have hotends without extruders (motor).
I tried to modify the code without success.
(I know that I could use the bed pins but actually I need more hotends)

DerAndere1, 2020-12-21:
I could have a look at this, but no promises. could you in turn test my 6-axis-Marlin (https://github.com/DerAndere1/Marlin/tree/6axis_PR1  ) with your setup? It supports endstops and auto-homing on 6 axes (X,Y,Z,A,B,C). Feedback is very welcome.

Daniele-Dondi, 2020-12-21:
Hi Derandere,
 I'm nice to speak with you. Actually I'm using your experimental version
marlin4pipetbot with 6 axis plus endstops and the only way to have one
hotend is to use the heated bed. In fact with 0 extruders the hotend part
is not compiled. I tried to modify the code with no success.
I'll try the 6axis version if you want a feedback

DerAndere1, 2020-12-21:
@Daniele-Dondi : Then stay on your current firmware for now. The link I gave you only has some more cleanup in the code but is otherwise the same firmware as you are already using. If you don't use it for research, could you attach your Configuration.h and Configuration_adv.h to a comment so I can see what features you have enabled? thanks

Daniele-Dondi, 2020-12-21:
Hi, I'm trying it to use for research, so I can attach the files I modified.
More or less all the lines I modified contains my name (//Dondi)

Daniele-Dondi, 2021-01-05:
Here it is my configuration:

_Hardware:_
Board: Megatronics v.3
Thermocouple (in the first connector)
6 motors for control syringes
6 endstops (1 for each syringe) (In detail: for X,Y,Z motors the X,Y,Z MIN pins were used; for I,J,K the X,Y,Z MAX plugs were used instead)
(many, 4 in the current version) servo controls (for valve opening and other stuff) 

_Software:_
The board pin file was modified to have 5 non extruder axes + 1 extruder OR 6 non extruder axes ONLY. All other configuration cannot work.
In the future pins reserved for the LCD control will be used for driving additional servos.
The bed temperature control uses the first thermocouple and the first heater cartridge output (this is used to control an external heater)

_Whitelist:_
(tested and working commands [for me these are enough for my purposes])
G1, G0, G28
M119, M140, M280, M400

My wish is to have available at least another PID control for a second heater

DerAndere1, 2021-01-05:
Thanks for the feedback. Regarding usage of hotends without extruder steppers: You should be able to set a dummy-value for the E-stepper pins and leave that dummy pin connected to nothing . E.g, If you remove the LCD display, in your pins_MEGATRONICS_3.h file, you could define 

```
#define E0_STEP_PIN                          14
#define E0_DIR_PIN                            14
#define E0_ENABLE_PIN                     14
``` 

In your Configuration.h file, define `EXTRUDERS 1` and define the correct `TEMP_SENSOR_0` value for your hotend. Changing EXTRUDERS 0 to EXTRUDERS 1 means that you will have to add a dummy-value fat the end of some arrays in Configuration.h, such as DEFAULT_AXIS_STEPS_PER_UNIT, DEFAULT_MAX_FEEDRATE, DEFAULT_MAX_ACCELERATION. Finally, connect your temperature sensors to the defined TEMP_SENSOR pins and connect your heater to the HEATER_0 pin and try if it works. I opened a similar issue at https://github.com/DerAndere1/Marlin/issues/42 where we can discuss the details.