MarlinFirmware/Marlin #20428: "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements, distinct e factors is enabled

FM101010, 2020-12-10:
### Bug Description

I have distinct e factors enabled however compile returns
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2766:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2766 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");

### Configuration Files

This is for a 400mm corexy using a(big tree tech) BTT 1.4 board and a BTT tft3.5 touch screen. 5 TMC2209 drivers, single e3d extruder.
This printer has 3 z motors running on 2 stepper drivers, the BTT 1.4 has dual output for the Z( I am using this for 2 of the Z stepper motors) Then I am using E1 as my Z2 stepper ( the 3rd Z motor is connected here) E0 is the extruder motor

### Steps to Reproduce

1. add extra e step factors after enabling distinct e factors
2. add extra acceleration and feed rate factors
3. compile

**Expected behavior:**

my code compiles

**Actual behavior:**

the compile fails due to 
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2766:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2766 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");

#### Additional Information

I was able to to compile successfully if I go into sanity check and comment out the lines that are throwing errors 
//static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");

[configh-configadv-fm.zip](https://github.com/MarlinFirmware/Marlin/files/5675463/configh-configadv-fm.zip)

ellensp, 2020-12-10:
You have #define EXTRUDERS 1 with #define DISTINCT_E_FACTORS 
Its even in the eg
  X, Y, Z, E0 [, E1[, E2...]]   (ie additional E elements only) 
you cannot and do not set steps/mm for additional Z stepper drivers. they all use the one Z settings.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-10 by ellensp

FM101010, 2020-12-10:
wow,
so i guess i better hit the road