MarlinFirmware/Marlin #20429: [BUG] SKR V1.4 Turbo and MAX31855 in Hardware SPI does not work when using RX0 (P0_03) as CS PIN

GadgetAngel, 2020-12-10:
### Bug Description

Using a SKR V1.4 Turbo (LPC1769) and a MAX31855 from Adafruit with a K-Type Thermocouple in Hardware SPI mode (EXP2 connector) the MIN_TEMP error comes up immediately.

### Configuration Files

[Configuration MAX31855_HW_SPI.zip](https://github.com/MarlinFirmware/Marlin/files/5675450/Configuration.MAX31855_HW_SPI.zip)

### Steps to Reproduce

1. SKR V1.4 Turbo board, use 3 UART stepper motor drivers in X, Y and Z. Use one SPI stepper motor driver in E0
2. With REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER on EXP1 and EXP2 use a snap on tap to access the following pins on EXP2 connector (P0_15 is you SCK_PIN) and (P0_17 is your MISO_PIN) as long as in pins_BTT_SKR_V1_4.h has `#define SDCARD_CONNECTION                  LCD`
3. now use the empty TFT connector to hook up the CS line from the MAX31855 board to P0_03 (RX0) pin
4. ensure that in configuration.h you disable the `//#define SERIAL_PORT_2 0` the secondary SERIAL_PORT_2.
5. in pins_BTT_SKR_common.h set the following:
```
#define TEMP_0_PIN                    P0_03
```
6. in configuration.h set the following:
```
//#define SERIAL_PORT_2 0
#define EXTRUDERS 1
#define TEMP_SENSOR_0 -3  
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
//#define SDSUPPORT
```
7. in configuration_adv.h set the following:
```
#define THERMOCOUPLE_MAX_ERRORS 15
#define SHOW_TEMP_ADC_VALUES
#define TMC_USE_SW_SPI
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
```
8. in pins_BTT_SKR_V1_4.h ensure the following:
```
//
// SD Connection
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION                  LCD
#endif
```

**Expected behavior:**

I expect my printer to use the temperature readings from the thermocouple and NOT to get and MIN TEMP error on boot up.

**Actual behavior:**

I get a MIN TEMP error! on bootup of the SKR V1.4 Turbo board.

#### Additional Information

When I use the MAX31855 **library call for hardware SPI** everything works fine.  I will add the hardware SPI call in my PR to fix this problem on the LPC178x boards.

ellensp, 2020-12-11:
you have #define TEMP_0_PIN                    P0_03   then are re using TEMP_0_PIN as various SS pins
I suspect something is seeing TEMP_0_PIN and setting it up incorrectly
I would suggest resetting TEMP_0_PIN back to P0_24_A1 and use a unique unused define name for the SS pin  if you wish to use it in a number of places

rhapsodyv, 2020-12-11:
Most of the boards aren't not prepared to used any spare pin for anything.... By default, the pins should be used for their planned function... 

The platform framework have some defaults that it load at startup, even if the pin is not being using inside Marlin. We can tune the framework to not do it, as the work done by for @sjasonsmith for GTR - he fine tuned the GTR variant, so the framework don't mess-up with non used pins. But it's not true for most of the other boards.

For LPC framework, we need to check if is marlin that is messing that PIN, or if the is the LPC framework that is doing some standard setup. Maybe @p3p can tell it by head.

But I will just start to work on this when my MAX31855 arrive. Until there, I cannot test anything, so it would be too hard to find any reason for this problems.

GadgetAngel, 2020-12-11:
@ellensp I will try your idea out and see if that solves the issue.

@rhapsodyv on the SKR V1.3 or SKR V1.4 or SKR V1.4 Turbo there are NOT a lot of free pins to use.  

So if a Marlin user wants to connect up additional equipment the options **are very limited**.  

I am trying to allow the Marlin user for LPC boards access to SPI temperature boards so that they can hook up a PT100 or K-Type Thermocouple as a temperature sensor.  In the forums I see a lot of question concerning this issue.  I have been asked **to write up a guide for the LPC boards (SKR V1.3, SKR V1.4, SKR V1.4 Turbo)** so that the Marlin user can have access to a PT100/K-Type thermocouple on a 3.3VDC LPC board.  The easiest and safest way to do this is via SPI protocol.

If you look at the boards and ask yourself what way could the user hook up these devices without having to hack away at the actual hardware on the LPC board then you will see that using the AUX-1/TFT connector signal pins and utilizing the hardware SPI protocol is by far the easiest ways.  

I believe using the AUX-1 connector is the most common way to add a temperature sensor to these LPC boards (Especially If the user wants to add two PT100s {one for each extruder}). The pins on the AUX-1 connector are the same pins on the TFT connector for the SKR V1.3 LPC board.  For the SKR V1.4 and SKR V1.4 Turbo there are additional pins but I am trying to find **a common way for the following three LPC boards:  SKR V1.3, SKR V1.4 and SKR V1.4 Turbo**.  

Since I want a common method for the SKR V1.3, SKR V1.4 and SKR V1.4 Turbo I am limited to the board with the least options which is the SKR V1.3 board.  Therefore, the SKR V1.3 only has two options for connecting up additional equipment: 1) if the user wants TWO additional temperature sensor because they have two extruders then they would use AUX-1 which has two pins but they are RX0 and TX0 with hardware SPI; 2) if the user wants only ONE additional temperature sensor then using the empty E1 driver socket by utilizing Software SPI protocol is the easiest way.

So these are the two methods I am testing on the LPC boards.

I have a work-a-round that I finished testing this morning.  I believe the problem resides in the SCLK pulse width the LPC framework produces for the SPI protocol.  It barely meets the SPI specs which are listed on each of the SPI board's data sheet (MAX31865/MAX31855/MAX6675 data sheets).  

My point: 
1) If the problem DOES NOT reside in the SCLK pulse width produced for the SPI protocol by the LPC framework then why would going to another library call allow the PIN to work as expected?

2) If Marlin's current LPC framework is "tying up the PIN" then **moving to a library call would not work** because Marlin would still have that PIN "tied up". But in actual testing when I do use a library call everything works as expected.

@rhapsodyv You are the expert here, not me.  I am a instrumentation engineer. I am good at testing things and I specialized in hardware to software interfaces but I am always still learning.  If I am wrong about my speculation then I will admit it.  All I need really is to understand what is causing the issue.  I can always find a way to make things work.  So I am offering to make all these issues work in Marlin on the LPC boards until you (@rhapsodyv) can **actually fix the SPI protocol on the LPC framework properly**.  If you rather I not submit the PR which will fix these SPI problems tell me now so I save my time and effort.

rhapsodyv, 2020-12-11:
@GadgetAngel I was just trying to say something about what could be messing with that pin.

I think your problems with those sensors lie in 3 main causes:

1) for hw spi, the marlin code can me sharing settings with other devices, making one mess with other. This is a ongoing fix for all marlin spi devices.

2) for hw spi using a pin that is messed up by other part of the code (what I commented before). In this case, even the bootloader could be messing with the pin. I need to take a careful look.

3) for sw spi, I think the code for max sensor on marlin is *very* old, as they had even c++ compilation error. That error was recently fixed, but it wasn't even compiling for a few months... So, it's probably broken for a lot of use cases. I need to review it.

About your PR: I will never discourage you. By the contrary: if you found a solution, It would be great to receive and merge it.

It's just I really can't test it now. In some weeks (2 or 3) my sensors will arrive and I can work on it.

GadgetAngel, 2021-01-12:
> This issue has had no activity in the last 30 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 7 days.

This issue is still under review because my pull request is still under review

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by GadgetAngel