MarlinFirmware/Marlin #20432: [BUG] TEMP_CHAMBER_HYSTERESIS

Lanwolf81, 2020-12-10:
### Bug Description

SKR 1.4 Turbo
Configuration_adv.h

#if TEMP_SENSOR_CHAMBER
  #define CHAMBER_MINTEMP             5
  #define CHAMBER_MAXTEMP            60
  #define TEMP_CHAMBER_HYSTERESIS     5   // (°C) Temperature proximity considered "close enough" to the target
  //#define CHAMBER_LIMIT_SWITCHING
  #define HEATER_CHAMBER_PIN       P1_00   // Chamber heater on/off pin
  //#define HEATER_CHAMBER_INVERTING false

#define TEMP_CHAMBER_HYSTERESIS  did not work.
Starting Heating at 1°C under the set temp.
I use M502 to reset the eeprom after flashing, but nothing change

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5676016/Marlin.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Change the  #define TEMP_CHAMBER_HYSTERESIS  over 1
Flash the new firmware
Test

**Expected behavior:**
Set the chamber temrature to 40°C. Marlin heating the chamber up to it. After the tempreatur is falling under 35°C start heating again.

**Actual behavior:**

Set the chamber temrature to 40°C. Marlin heating the chamber up to it. After the tempreatur is falling under 39°C start heating again.

#### Additional Information

sjasonsmith, 2020-12-22:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

If it is still an issue please provide updated configuration files which can build with `bugfix-2.0.x`.

sjasonsmith, 2020-12-31:
Closing, since the original submitter is unresponsive. @Lanwolf81 if you test with `bugfix-2.0.x` and still have this issue this can be re-opened.

closed 2020-12-31 by sjasonsmith