MarlinFirmware/Marlin #20434: [BUG] mks LCD12864a. White Screen

Jeong-Hyung-min, 2020-12-11:
I am using MEGA2560 Board and RAMPS 1.6 and mks LCD12864a. The firmware is 1.1.9.1
I am satisfied with the mks LCD12864a.
It works fine except for one thing.
When the power is turned on and booted, the screen of mks LCD12864a is all white, making it impossible to distinguish characters. 
[image]

If the 3D printer is turned on and off three or four times, it works normally. The connecting cable is about 15cm.
[image]

Could it be a timing issue? Please tell me the solution.

thisiskeithb, 2020-12-11:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

mks-viva, 2020-12-16:
@Jeong-Hyung-min 
Hi! I would recommend you to use the latest version of marlin 2.x firmware

sjasonsmith, 2020-12-16:
Try re-initializing your EEPROM through the menus or using M502 then M500.

Screens with software-configurable contrast will often have bad contrast values saved in the EEPROM if you have just added the screen to your configuration.

sjasonsmith, 2020-12-16:
I missed the firmware version you listed. As others have said you will need to update to Marlin 2.x to support this screen properly.

sjasonsmith, 2020-12-20:
Given that this is for 1.x firmmare, I am closing this. We no longer develop for 1.x, and changes for this display will not be back-ported to it.

closed 2020-12-20 by sjasonsmith