MarlinFirmware/Marlin #20435: [BUG] dual extruder setup needs workaround when using g29 before a print

frizzle101101, 2020-12-11:
Most slicers perform an active extruder heat up at the beginning of a print, the problem im facing is i have a dual extruder setup and i simply want to print with the secont hotend or tool 1 but when the g29 command is run it automatically switches the active extruder back to tool 0 as shown in the following log, this inturn tries to print through the wrong nozzle.

`
Recv: echo:busy: processing
Recv:  T:137.67 /0.00 B:60.27 /60.00 T0:137.67 /0.00 T1:202.01 /200.00 @:0 B@:32 @0:0 @1:0
Recv: echo:busy: processing
Recv:  T:136.67 /0.00 B:60.14 /60.00 T0:136.67 /0.00 T1:201.77 /200.00 @:0 B@:39 @0:0 @1:0
Recv: echo:Active Extruder: 1
Recv: X:0.00 Y:-6.00 Z:0.00 E:83.41 Count X:0 Y:-1920 Z:0
Recv: ok
Send: N11 G29*34
Recv:  T:135.70 /0.00 B:60.20 /60.00 T0:135.70 /0.00 T1:201.46 /200.00 @:0 B@:33 @0:0 @1:0
Recv: echo:Active Extruder: 0
Recv: echo:busy: processing
Recv:  T:134.77 /0.00 B:60.08 /60.00 T0:134.77 /0.00 T1:200.62 /200.00 @:0 B@:40 @0:0 @1:9
Recv: echo:busy: processing
Recv:  T:133.88 /0.00 B:60.10 /60.00 T0:133.88 /0.00 T1:199.80 /200.00 @:0 B@:37 @0:0 @1:17
Recv: echo:busy: processing
Recv:  T:132.97 /0.00 B:60.14 /60.00 T0:132.97 /0.00 T1:198.86 /200.00 @:0 B@:32 @0:0 @1:27
Recv: echo:busy: processing
Recv:  T:131.64 /0.00 B:59.99 /60.00 T0:131.64 /0.00 T1:198.27 /200.00 @:0 B@:39 @0:0 @1:32
Recv: echo:busy: processing
Recv:  T:130.52 /0.00 B:60.07 /60.00 T0:130.52 /0.00 T1:197.81 /200.00 @:0 B@:32 @0:0 @1:37
Recv: echo:busy: processing
Recv:  T:129.58 /0.00 B:59.99 /60.00 T0:129.58 /0.00 T1:197.93 /200.00 @:0 B@:37 @0:0 @1:35
Recv: echo:busy: processing
Recv:  T:128.76 /0.00 B:59.92 /60.00 T0:128.76 /0.00 T1:198.21 /200.00 @:0 B@:41 @0:0 @1:32
Recv: echo:busy: processing
Recv:  T:128.02 /0.00 B:59.93 /60.00 T0:128.02 /0.00 T1:199.03 /200.00 @:0 B@:37 @0:0 @1:23
Recv: echo:busy: processing
Recv:  T:127.25 /0.00 B:59.80 /60.00 T0:127.25 /0.00 T1:199.86 /200.00 @:0 B@:46 @0:0 @1:15
Recv:  T:126.35 /0.00 B:59.83 /60.00 T0:126.35 /0.00 T1:200.76 /200.00 @:0 B@:45 @0:0 @1:5
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:125.38 /0.00 B:59.92 /60.00 T0:125.38 /0.00 T1:201.49 /200.00 @:0 B@:32 @0:0 @1:0
Recv: echo:busy: processing
Recv:  T:124.53 /0.00 B:59.86 /60.00 T0:124.53 /0.00 T1:201.84 /200.00 @:0 B@:37 @0:0 @1:0
Recv:  T:123.72 /0.00 B:60.03 /60.00 T0:123.72 /0.00 T1:201.94 /200.00 @:0 B@:25 @0:0 @1:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:122.93 /0.00 B:59.80 /60.00 T0:122.93 /0.00 T1:201.49 /200.00 @:0 B@:45 @0:0 @1:1
Recv: echo:busy: processing
Recv:  T:122.21 /0.00 B:60.15 /60.00 T0:122.21 /0.00 T1:200.83 /200.00 @:0 B@:21 @0:0 @1:9
Recv: echo:busy: processing
Recv:  T:121.46 /0.00 B:60.13 /60.00 T0:121.46 /0.00 T1:200.00 /200.00 @:0 B@:25 @0:0 @1:18
Recv: echo:busy: processing
Recv:  T:120.66 /0.00 B:59.97 /60.00 T0:120.66 /0.00 T1:199.43 /200.00 @:0 B@:39 @0:0 @1:22
Recv: echo:busy: processing
Recv:  T:119.72 /0.00 B:60.06 /60.00 T0:119.72 /0.00 T1:198.64 /200.00 @:0 B@:32 @0:0 @1:31
Recv: echo:busy: processing
Recv:  T:118.86 /0.00 B:59.86 /60.00 T0:118.86 /0.00 T1:198.44 /200.00 @:0 B@:48 @0:0 @1:31
Recv:  T:118.06 /0.00 B:60.03 /60.00 T0:118.06 /0.00 T1:198.27 /200.00 @:0 B@:34 @0:0 @1:33
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:117.21 /0.00 B:59.95 /60.00 T0:117.21 /0.00 T1:198.72 /200.00 @:0 B@:42 @0:0 @1:27
Recv: echo:busy: processing
Recv:  T:116.51 /0.00 B:59.93 /60.00 T0:116.51 /0.00 T1:199.06 /200.00 @:0 B@:43 @0:0 @1:24
Recv:  T:115.70 /0.00 B:60.01 /60.00 T0:115.70 /0.00 T1:199.66 /200.00 @:0 B@:36 @0:0 @1:18
Recv: echo:busy: processing
Recv:  T:114.92 /0.00 B:60.09 /60.00 T0:114.92 /0.00 T1:200.38 /200.00 @:0 B@:28 @0:0 @1:11
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:113.86 /0.00 B:60.06 /60.00 T0:113.86 /0.00 T1:201.18 /200.00 @:0 B@:32 @0:0 @1:3
Recv: echo:busy: processing
Recv:  T:113.12 /0.00 B:60.60 /60.00 T0:113.12 /0.00 T1:201.60 /200.00 @:0 B@:0 @0:0 @1:0
Recv: echo:busy: processing
Recv:  T:112.12 /0.00 B:60.14 /60.00 T0:112.12 /0.00 T1:201.39 /200.00 @:0 B@:30 @0:0 @1:2
Recv:  T:111.34 /0.00 B:60.13 /60.00 T0:111.34 /0.00 T1:200.80 /200.00 @:0 B@:30 @0:0 @1:10
Recv: echo:busy: processing
Recv:  T:110.59 /0.00 B:60.16 /60.00 T0:110.59 /0.00 T1:200.21 /200.00 @:0 B@:27 @0:0 @1:16
[... 132 lines cut ...]
Send: N44 M108*26
Recv: ok
Send: N45 M84*46
Recv:  T:99.70 /0.00 B:60.00 /60.00 T0:99.70 /0.00 T1:198.55 /200.00 @:0 B@:32 @0:0 @1:32
Recv: echo:busy: processing
Recv:  T:99.01 /0.00 B:59.90 /60.00 T0:99.01 /0.00 T1:198.64 /200.00 @:0 B@:39 @0:0 @1:30
Recv: ok
Send: N46 M104 T0 S0*19
Recv: ok
Send: N47 M104 T1 S0*19
Recv: ok
Send: N48 M140 S0*89
Recv: ok
Send: N49 M106 S0*90
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv:  T:98.40 /0.00 B:59.96 /0.00 T0:98.40 /0.00 T1:199.01 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:97.74 /0.00 B:59.88 /0.00 T0:97.74 /0.00 T1:199.32 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:97.15 /0.00 B:59.91 /0.00 T0:97.15 /0.00 T1:199.35 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:96.55 /0.00 B:59.71 /0.00 T0:96.55 /0.00 T1:199.49 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:96.02 /0.00 B:59.53 /0.00 T0:96.02 /0.00 T1:199.12 /0.00 @:0 B@:0 @0:0 @1:0
Recv:  T:95.53 /0.00 B:59.30 /0.00 T0:95.53 /0.00 T1:198.61 /0.00 @:0 B@:0 @0:0 @1:0
`

if g29 is taken out of the print it works fine and there is no issues, the g29 command should not change the active tool back to tool 0.

sjasonsmith, 2020-12-18:
This is missing key information which is required in the template when filing a bug report. Please update the description to follow the new issue template, or this is likely to be closed.

You can find the template by clicking the button to report a new bug, although please do not actually submit the new one, update it here instead.

sjasonsmith, 2020-12-20:
Closed due to failure to follow instructions. There are no configuration files, steps to reproduce, etc.

closed 2020-12-20 by sjasonsmith