MarlinFirmware/Marlin #20436: Probably not a (BUG)Anet A8 Anet v1.5 Board ZONESTAR LCD 2004 Buttons Not Working (2.0 Bugfix)

tamedkek, 2020-12-11:
### Bug Description
Using the anet board with a zone star LCD 2004 display, only the center button works. I can only go from the info screen to options but I can't navigate up or down. I have 2 different LCD 2004 displays, I have tried with both, wiggled the cable, switched connector sides, and reset many times.
### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5677441/Configs.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce
Enable ZONESTAR_LCD
Clean last environment
Build (auto-build marlin)
Upload
Try and load SD card
**Expected behavior:**
I can use the down button to load files from my SD

**Actual behavior:**
I can click the middle button on the info screen, all other buttons don't work, can still print through USB and confirm start on the screen.

-----------------------------------------------------------

No buttons being pressed
22:55:16.963 -> PIN:  10   Port: D2 (A21)  LCD_PINS_D4                            Output = 0  
22:55:16.963 -> PIN:  11   Port: D3 (A20)  LCD_PINS_D5                            Output = 0  

22:55:16.997 -> PIN:  15   Port: D7 (A16)  X_STEP_PIN                             protected 
22:55:16.997 -> PIN:  16   Port: C0 (A15)  LCD_PINS_D6                            Output = 0  
22:55:16.997 -> PIN:  17   Port: C1 (A14)  LCD_PINS_D7                            Output = 0  

22:55:17.100 -> PIN:  27   Port: A4 (A 4)  SERVO0_PIN                             Input  = 0  
22:55:17.100 -> PIN:  28   Port: A3 (A 3)  LCD_PINS_RS                            Output = 1  
22:55:17.100 -> .                          LCD_SDSS                               Output = 1  
22:55:17.100 -> PIN:  29   Port: A2 (A 2)  LCD_PINS_ENABLE                        Output = 0  
22:55:17.134 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =  1021  
22:55:17.134 -> PIN:  31   Port: A0 (A 0)  SDSS                                   Output = 1  
22:55:17.134 -> ok

Middle
22:57:24.627 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =   321  

Left
22:57:52.473 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =    91  

Bottom
22:58:19.156 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =   176  

Right
22:58:40.388 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =   509  

Up
22:59:04.016 -> PIN:  30   Port: A1 (A 1)  ADC_KEYPAD_PIN                         Analog in =   691

frontrunnersoftware, 2020-12-16:
I can confirm ZONESTAR_LCD buttons are not working in Bugfix 2.0.x, yet work fine in release version 2.0.7.2

ellensp, 2020-12-17:
Using a test setup I was able to confirm that 2.0.7.2 works as expected, but bugfix only the center button works (sort of)

ellensp, 2020-12-17:
Seem to have been broken in commit https://github.com/MarlinFirmware/Marlin/commit/f6ffbe548c81abfeecc996c8f2bffd27cabe0014
Sadly this is a huge commit.

ellensp, 2020-12-17:
Error is in Marlin/src/lcd/marlinui.cpp
`if (TERN0(REPRAPWORLD_KEYPAD, handle_keypad()))` 
should be 
`if (TERN0(IS_RRW_KEYPAD, handle_keypad()))`

ellensp, 2020-12-17:
I don't even have this type of CD, had to make my own compatible one. 
[image]

ellensp, 2020-12-17:
This has been merged into bugfix. closing

closed 2020-12-17 by ellensp