MarlinFirmware/Marlin #20441: Anet a8 plus on TMC 2208 drivers make a squeak, and do not move

y3kov, 2020-12-11:
Anet a8 plus on SKR v1.4 TMC 2208 drivers / Bought on Anet A8 plus Anet a8 plus on TMC 2208 v2. 0 drivers tell me how to put jumpers under the drivers on the Board? I tried many configurations for the marlin firmware. But in some cases, it doesn't compile. With a successful compilation after flashing, the Axes move by 1 mm and start to make a squeak, and do not move
. If you lower the speed of movement, the engines make a lot of noise but move. When Parking the Z-axis reaches the zero point, it rests on the button and stops, there is no rollback.Please help me, when rearranging jumpers in the right row, I tried to lower the voltage, but it did not give any results!
By the way, the driver on the Z axis is very hot (I did all the tests on the Z axis)

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/5680819/Conf.zip)

thisiskeithb, 2020-12-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-11 by thisiskeithb

y3kov, 2020-12-11:
With this configuration, everything moves as it should, perfectly. But there is one but! When the carriage is moved by 1 cm programmatically via the printer control when measuring, it is actually moved by 1.3 cm. A 2x2x2 cube turns out to be X2. 4 Y2. 4 Z2. 0
[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/5681070/Conf.zip)

y3kov, 2020-12-11:
After all, I correctly addressed the problem in the firmware!

> Эта очередь задач предназначена для отчетов об ошибках Marlin и проблем, связанных с разработкой, и мы предпочитаем не рассматривать здесь вопросы поддержки пользователей. (Как указано на [этой странице](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question) .) Для получения наилучших результатов при получении справки по настройке и устранению неполадок используйте следующие ресурсы:
> 
> * MarlinFW.org [Документация Marlin](http://marlinfw.org)
> * RepRap.org [Марлин Форум](http://forums.reprap.org/list.php?415)
> * [Форумы Tom's 3D](https://forum.toms3d.org/)
> * Группа Facebook ["Прошивки Марлина"](https://www.facebook.com/groups/1049718498464482/)
> * Группа Facebook [«Прошивки Marlin для 3D-принтеров»](https://www.facebook.com/groups/3Dtechtalk/)
> * [Конфигурация Marlin](https://www.youtube.com/results?search_query=marlin+configuration) на YouTube
> * Сервер Marlin Discord. Ссылка для присоединения: https://discord.gg/n5NJ59y
> 
> После обращения за помощью к сообществу, если консенсус указывает на ошибку в Marlin, вы должны опубликовать [отчет](https://github.com/MarlinFirmware/Marlin/issues/new/choose) об [ошибке](https://github.com/MarlinFirmware/Marlin/issues/new/choose) .

After all, I correctly addressed the problem in the firmware!