MarlinFirmware/Marlin #20442: Probing Failed: Probe Temperature Compensation SKR1.3

floriankogler, 2020-12-11:
I use Marlin 2.0.7.2 release and a fix mounted inductive probe.

On G76 or G76 P the probing fails always during the second measurement.
On G76 the bed calibration process works fine, then it fails during second measurement of probe calibration process.

I get always the same error massage: Error: Probing Failed!

Here the thermal probe compensation section in configuration_adv.h

```
/**
 * Thermal Probe Compensation
 * Probe measurements are adjusted to compensate for temperature distortion.
 * Use G76 to calibrate this feature. Use M871 to set values manually.
 * For a more detailed explanation of the process see G76_M871.cpp.
 */
#if HAS_BED_PROBE && TEMP_SENSOR_PROBE && TEMP_SENSOR_BED
  // Enable thermal first layer compensation using bed and probe temperatures
  #define PROBE_TEMP_COMPENSATION

  // Add additional compensation depending on hotend temperature
  // Note: this values cannot be calibrated and have to be set manually
  #if ENABLED(PROBE_TEMP_COMPENSATION)
    // Park position to wait for probe cooldown
    #define PTC_PARK_POS   { 5, 5, 100 }

    // Probe position to probe and wait for probe to reach target temperature
    #define PTC_PROBE_POS  { 90, 100 }

    // Enable additional compensation using hotend temperature
    // Note: this values cannot be calibrated automatically but have to be set manually
    //#define USE_TEMP_EXT_COMPENSATION

    // Probe temperature calibration generates a table of values starting at PTC_SAMPLE_START
    // (e.g. 30), in steps of PTC_SAMPLE_RES (e.g. 5) with PTC_SAMPLE_COUNT (e.g. 10) samples.

    //#define PTC_SAMPLE_START  30.0f
    //#define PTC_SAMPLE_RES    5.0f
    //#define PTC_SAMPLE_COUNT  10U

    // Bed temperature calibration builds a similar table.

    #define BTC_SAMPLE_START  50.0f
    //#define BTC_SAMPLE_RES    5.0f
    #define BTC_SAMPLE_COUNT  14U

    // The temperature the probe should be at while taking measurements during bed temperature
    // calibration.
    //#define BTC_PROBE_TEMP 30.0f

    // Height above Z=0.0f to raise the nozzle. Lowering this can help the probe to heat faster.
    // Note: the Z=0.0f offset is determined by the probe offset which can be set using M851.
    //#define PTC_PROBE_HEATING_OFFSET 0.5f

    // Height to raise the Z-probe between heating and taking the next measurement. Some probes
    // may fail to untrigger if they have been triggered for a long time, which can be solved by
    // increasing the height the probe is raised to.
    #define PTC_PROBE_RAISE 20U

    // If the probe is outside of the defined range, use linear extrapolation using the closest
    // point and the PTC_LINEAR_EXTRAPOLATION'th next point. E.g. if set to 4 it will use data[0]
    // and data[4] to perform linear extrapolation for values below PTC_SAMPLE_START.
    //#define PTC_LINEAR_EXTRAPOLATION 4
  #endif
```

thisiskeithb, 2020-12-12:
Please edit your post and attached your configs as a .zip as outlined in the instructions and remove them from inline.

Also: Download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

floriankogler, 2020-12-12:
I just updated my post and added some information.
I still have the issue with the latest bugfix.

TheNeskik, 2020-12-15:
You can try to lower your probe. When you calibrate the probe, each point will keep in mind the temperature "drift" so the probe will try to trigger higher than it currently can.

So : 
1 - Lower your probe
2 - Tune your z-offset with a bed at 60°C and your probe at 30 °C
3 - Run G76 P

See if it helps :)

floriankogler, 2020-12-15:
So I tried to lower the probe, and tuned my offset. I had a z-offset of -2.489, if that is of ralevance for you.

But unfortunately the error occured during first probing.

TheNeskik, 2020-12-15:
Are you able to perform a bed level without temperature compensation ? If you use a PINDA V2 you should try to lower your probe in order to get an offset around 1.00 mm above your nozzle.
I don't think your problem is marlin. I advice you to come on the Discord :) if we point that there are no problem with your probe configuration. We could consider that it can be a bug :)

floriankogler, 2020-12-15:
Thank you for your help, I was able to find a solution.

In the file Marlin\src\gcode\calibrate\G76_M192_M871.cpp there is the following section:
````
auto g76_probe = [](const TempSensorID sid, uint16_t &targ, const xy_pos_t &nozpos) {
    do_z_clearance(5.0); // Raise nozzle before probing
    const float measured_z = probe.probe_at_point(nozpos, PROBE_PT_STOW, 0, false, false);  // verbose=0, probe_relative=false, sanity_check=false (added by florian kogler)
    if (isnan(measured_z))
      SERIAL_ECHOLNPGM("!Received NAN. Aborting.");
    else {
      SERIAL_ECHOLNPAIR_F("Measured: ", measured_z);
      if (targ == cali_info_init[sid].start_temp)
        temp_comp.prepare_new_calibration(measured_z);
      else
        temp_comp.push_back_new_measurement(sid, measured_z);
      targ += cali_info_init[sid].temp_res;
    }
    return measured_z;
  };
````
In the line ````const float measured_z = probe.probe_at_point(nozpos, PROBE_PT_STOW, 0, false, false);```` is the last parameter (sanity check) in the original file left out. Due to the default argument it will be set to true.

So, it triggers an "Probe fail! - triggered early" due to the temperature "drift" of the probe.
As you can see in Marlin\src\module\probe.cpp:
````
float Probe::run_z_probe(const bool sanity_check/*=true*/) {
  DEBUG_SECTION(log_probe, "Probe::run_z_probe", DEBUGGING(LEVELING));

  auto try_to_probe = [&](PGM_P const plbl, const float &z_probe_low_point, const feedRate_t fr_mm_s, const bool scheck, const float clearance) {
    // Do a first probe at the fast speed
    const bool probe_fail = probe_down_to_z(z_probe_low_point, fr_mm_s),            // No probe trigger?
               early_fail = (scheck && current_position.z > -offset.z + clearance); // Probe triggered too high?
    #if ENABLED(DEBUG_LEVELING_FEATURE)
      if (DEBUGGING(LEVELING) && (probe_fail || early_fail)) {
        DEBUG_PRINT_P(plbl);
        DEBUG_ECHOPGM(" Probe fail! -");
        if (probe_fail) DEBUG_ECHOPGM(" No trigger.");
        if (early_fail) DEBUG_ECHOPGM(" Triggered early.");
        DEBUG_EOL();
      }
    #else
      UNUSED(plbl);
    #endif
    return probe_fail || early_fail;
  };
````

So, I changed the parameter to false and now the calibration runs without a problem.

closed 2021-01-23 by github-actions[bot]