MarlinFirmware/Marlin #20445: [BUG] Using Baby Stepping from the LCD will cause the DELTA machine to halt.

yoch2015, 2020-12-12:
### Bug Description
 Using Baby Stepping from the LCD will cause the DELTA machine to halt.
 Marlin-bugfix-2.0.x. Downloaded on December 9th.
 Marlin-bugfix-2.0.7.1 has the same problem.

### Configuration Files
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/5682497/Configuration.h.zip)
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/5682498/Configuration_adv.h.zip)

### Step
1.Select Tune from the LCD and select Babystep Z.
2.The LCD will display Baby step Z: +00.000, turn the encoder to stop and restart the controller.
3.The following message is displayed in Octopi.
Communication timeout while printing, trying to trigger response from printer. 
Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: start

**Expected behavior:**
 Z changes by turning the encoder.

**Actual behavior:**
 Controller restart.

thisiskeithb, 2020-12-12:
Enable `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT` in your config.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sjasonsmith, 2020-12-12:
Can you try enabling `BABYSTEP_ALWAYS_AVAILABLE`, and babystep without a print in progress? I would like to know if that also causes the problem, since it will be an easier way to reproduce the issue.

yoch2015, 2020-12-12:
Thank you for your quick answer.
Even if BABYSTEP_ALWAYS_AVAILABLE is enabled, it will halt in the same way.

yoch2015, 2020-12-12:
Diffed babystep.cpp.
Line:58
New : if (DISABLED(BABYSTEP_WITHOUT_HOMING) && axes_should_home(_BV(axis))) return;
Old : if (DISABLED(BABYSTEP_WITHOUT_HOMING) && !TEST(axis_known_position, axis)) return;
Is it related?

closed 2021-02-04 by github-actions[bot]

yoch2015, 2021-03-12:
This issue is divided into:
8bit board (mega2560) - occered.
32bit board (LPC1768) - Doesn't occered.