MarlinFirmware/Marlin #20448: Compile error; Marlin\src\HAL\STM32F1  serial port, motherboard selection

xruell, 2020-12-12:
### Bug Description
When compiling I get errors related to Marlin\src\HAL\STM32F1  serial port, motherboard 

[JXC_XRUELL_E3v4.2.7.zip](https://github.com/MarlinFirmware/Marlin/files/5683465/JXC_XRUELL_E3v4.2.7.zip)
 

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sjasonsmith, 2020-12-12:
The problem is that serial port -1 is invalid for your board. Please refer to one of the example configurations for your board to find the correct port number.

-1 is for USB implemented inside your MCU. This board doesn’t use that, because it has an external serial to USB chip.

I thought we had better error messages than this. I will try building your configurations to see what output it generates.

xruell, 2020-12-12:
It would seem to be that easy, however, it does this when using the example config. I at one point forget what I just randomly put in that did not result in that error.  I am not using the serial port, yet. Still got the motherboard error.

FanDjango, 2020-12-12:
This can be closed. We're on discord and we're getting there. It's not a bug.

sjasonsmith, 2020-12-12:
Thanks @FanDjango, I was about to post my findings that my expected error is reported, at least with the latest bugfix comments.

I see the following error, which was recently added to help inform people their board won't work with the port set to -1.

`Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2370:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."`

closed 2020-12-12 by sjasonsmith

xruell, 2020-12-12:
Thank you guys!    Well, turns out I was just missing the proper info in #define MOTHERBOARD BOARD_(*********)  line in Configuration.h

Thank you for your help. I learned a bit more about PIO and how this setting works. 

In my defense it had been a month since I tried to update.