MarlinFirmware/Marlin #20449: Ender 3 v2 UBL 15x15 Mesh save using G29 S0 causes reboot/reset

kyraxyg, 2020-12-12:
### Bug Description

I have been using ABL 10x10 mesh for bed leveling. And a similar issue has been corrected by @sjasonsmith ( Bug:#20307)
I have switched to UBL with 15x15 mesh as my glass bed is heavily wrapped

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5683523/config.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. G29 P1
2. G29 P3 as long as all points are filled.
3. G29 S0 

**Expected behavior:**

Saving the mesh successfully.

**Actual behavior:**
Printer/board restarts

#### Additional Information

As mentioned a previous issue saving mesh to eprom using M500 was caused by watchdog and solved by @sjasonsmith . I think this might be a similar issue but only the save process is by G29 S0

Mezabaru, 2020-12-16:
you need to reduce your eeprom size + potentially ram usage!  I recommend disabling features you don't use!

sjasonsmith, 2020-12-16:
It is probably actually a watchdog timeout because writing this flash is very slow.

I put in some calls to the watchdog when writing a large ABL mesh, but did not test a large UBL mesh when I did that.

kyraxyg, 2020-12-16:
I don't think this is realted to eeprom size either as once the mesh with 15x15 saved I think this is realted to watchdog as in the previous issue. The only difference is the previous issue was watchdog being triggered while saving eeprom using M500. But in the UBL case the saving mech is done by G29 S0 or G29 S1 ... I also think this is realted to watchdog like the previous one.

sjasonsmith, 2020-12-19:
The following sequence seems to reliably trigger the reboot:

```
G29 P0
G29 Q2
G29 S0
G29 Q0
G29 S0
```

sjasonsmith, 2020-12-20:
@kyraxyg this should now be fixed in the `bugfix-2.0.x` branch. Please try it and out close it out and close this if it has solved your issue.

sjasonsmith, 2020-12-22:
I am presuming this has been fixed. Please let me know if you still have issues when you get around to testing it.

closed 2020-12-22 by sjasonsmith

kyraxyg, 2020-12-22:
Sorry for the delay. I didn't have the time to check. I have just compiled the new bugfix and will hopefully test today. It is ok to be closed. I will inform if the issue still persists.

PR #20510 (2020-12-20): Improve Creality EEPROM watchdog behavior
### Description

The BL24Cxx EEPROM is very slow, and can trigger watchdog resets while writing large meshes.

An earlier change improved this, but would only service the watchdog every 128th byte if (and only if) that byte was being modified. With some patterns of data this caused the watchdog to expire and reboots to still occur.

This change tracks the number of bytes actually written, so that it will service the watchdog for every 128th _written_ byte.

### Benefits

Avoids watchdog resets when storing large meshes.

### Configurations

Ender 3 Pro [... cut]