MarlinFirmware/Marlin #20461: [FR] Disable all steppers but extruder during filament change from main menu when there is no active print

mseddon, 2020-12-13:
### Description

Disable X, Y, Z (but obviously not extruders) during filament change when there is no active filament

### Feature Workflow

1. Cold boot a marlin printer
2. Select 'change filament', and preselect temp. Disable X/Y/Z steppers when it's possible to detect we're not mid-print
3. This allows for a nozzle that is flush against the build plate or similar to be easily manually corrected so you don't make even more of a mess than you intend.

#### Additional Information

So I just hit 'change filament' because I started a print but luckily (or not), the nozzle was at z=0.1. I then purged into that position, which caused buildup over the nozzle. I was swapping to the same material, and the same colour, so this is not an issue and the blob was fairly minor and easy to remove.  However, had I been switching materials, then I would need to purge far more, and this would have risked backing up into the nozzle/all over the tool head.  Is it possible to disable the X/Y/Z steppers so I can e.g. manually turn the z-rod while the purge is occurring in this case (menu activation, no print in progress).

If there is a safety issue with this (e.g. risk that a disabled stepper may 'drift') then perhaps allow a config to raise the Z by a preset amount before filament change from the menu, specifically when going from the menu manually when no print is active.

This is my 2 cents. Hopefully the problem makes sense, but I would defer to you the best solution for this use-case. Cheers.

sjasonsmith, 2020-12-16:
Disabling motors means you lose the machine's position. You would have to re-home all your axis to resume the print. That would be doable for X/Y (but might not be perfect), but is often impossible for Z.

mseddon, 2020-12-16:
Good shout. To be fair just raising the Z up by 20mm would be handy in my case.

tpruvot, 2020-12-16:
#define NOZZLE_PARK_Z_RAISE_MIN   2   // (mm) Always raise Z by at least this distance

thisiskeithb, 2020-12-16:
> #define NOZZLE_PARK_Z_RAISE_MIN 2 // (mm) Always raise Z by at least this distance

To expand on this, take a look at the entire [Nozzle Park section](https://github.com/MarlinFirmware/Marlin/blob/c6dcf9d0067fdd6c81f8f229dd18aa34ac84c90a/Marlin/Configuration.h#L1580-L1601) as well as [Advanced Pause](https://github.com/MarlinFirmware/Marlin/blob/c6dcf9d0067fdd6c81f8f229dd18aa34ac84c90a/Marlin/Configuration_adv.h#L2107-L2160).

Your feature request seems covered by existing park/raise/filament change settings within Marlin.

mseddon, 2020-12-25:
Thanks for the clarification, I will close this. Cheers!

closed 2020-12-25 by mseddon