MarlinFirmware/Marlin #20467: [BUG] FAN_SOFT_PWM doesn't work for controller fan.

Licoze, 2020-12-14:
### Bug Description

When #define **FAN_SOFT_PWM** is set seems controller (motherboard) fan is not managed by software.

Just to make sure that it's not a motherboard issue I changed **CONTROLLER_FAN_PIN** to -1 in _pins_BTT_SKR_MINI_E3_V2_0.h_ and disabled **USE_CONTROLLER_FAN**  so it became a fan 2 for hotend. Then I set speed for both fans to 50% and there wasn't any noise.
 
**Motherboard** : BTT SKR Mini E3 V2.0
### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5691289/Marlin.zip)

### Steps to Reproduce

1. Set FAN_SOFT_PWM
2. Upload firmware
3. Go to **Configuration** - > **Controller Fan** -> set **Idle Speed**  to any value in range 20-80%

**Expected behavior:**

There are no any fan PWM noise

**Actual behavior:**

Fan PWM noise

sjasonsmith, 2020-12-18:
I peeked through the code and I am pretty sure you are correct that the controller fan (and likely chamber fan) cannot be used with SOFT_PWM.

I'm not entirely sure whether to leave this as a bug or change it to a feature request. It might depend on how extensive the changes are to make this happen.

Licoze, 2020-12-18:
I guess until this behavior described in documentation- it’s a bug.

Bastlwastl84, 2020-12-20:
having same problem with a Bigtree SKR Mini E3 V2.0 and my Parcooling Fan runs only 0% or 100% and only starting if opening the Menu in Marlin Mode. Sometimes i need to set to 100%, than the fan starting

sjasonsmith, 2020-12-20:
@Bastlwastl84 that is not the same problem. Your issue can likely be solved through configurations. This report is specifically for the complete lack of Soft PWM support for controller fans.

I suggest you go talk to some people in one of our user forums for assistance debugging your issue. Multiple links to help resources are provided on the page you see when reporting a new issue:
https://github.com/MarlinFirmware/Marlin/issues/new/choose

sjasonsmith, 2020-12-22:
I suspect that FAN_SOFT_PWM may _only_ apply to actual part-cooling fans. Any AUTO_FANS, chamber fans, and controller fans use different code paths which bypass SOFT_PWM control.

I do not know how these are impacted by FAST_PWM_FAN.

A deeper look is warranted, and either the fan operations should be made consistent, or documentation (primarily Configuration.h comments) should be updated.

Licoze, 2020-12-23:
>should be made consistent
Agree, I would move SOFT_PWM to lower abstraction level.
I was trying to inspect the code from top to bottom but "define" - oriented programming is too hard to understand for me ;(

Johok1, 2021-01-01:
Disclaimer: I'm new to this repo! (But I have a good bit of time configuring Marlin). 

@Licoze If I understand correctly, you are saying that the controller fan does not go down to the software frequency (which is in turn supposed to make it quieter?) After reading through the documentation for a while this seems to be the intended feature but please correct me if I'm mistaken. 

It is decisively the case that the controller fan is separate from the other fans. The other fans seem to be controlled in Marlin/module/temperature... whereas the controller fan gets its operation from Marlin/features/controllerfan... However, the pin for the controllerfan is initialized in the temperature class (which seems to serve as the masterclass for all fan control so to speak).  

It probably sounds like im repeating things already established. I apologize, I am doing so in case I have made some misconception. If this is the case please correct me. 

Now, from what I understand the controllerfan is used for cooling down filament as it leaves a hotend. I am familiar with this process, the controllerfans speed can be adjusted if the filament comes out too hard and doesn't adhere to the bed or if it comes out too hot and burns. So, would it even be wise then to allow the controllerfan to be effected by the software pwm fan setting? If I understand it all, the software pwm setting sets the frequency of pwm signal going to the fan to some proportion of the signal going to the bed. If this is the case, wouldn't it effectively speed up or slow down the fan? If this is how it works, then that cannot apply to the controller fan as it must be at a certain preset speed in order for the filament to extrude and adhere properly. 

I have likely misunderstood something, and I hope this can get clarified!

Licoze, 2021-01-12:
Hi, @Johok1 
> Now, from what I understand the controllerfan is used for cooling down filament

Nope, I guess you're talking about hotend fan. 
Controller fan is used for cooling a motherboard.

Bastlwastl84, 2021-01-12:
Cooling the printed part.

Johok1, 2021-01-14:
Thanks for clarifying that. So then I guess the only issue here is that the fan speed can't be controlled and is making noise as a result?

closed 2021-02-20 by github-actions[bot]

Licoze, 2021-02-23:
Up.