MarlinFirmware/Marlin #20470: [BUG] LPC1769 high temp thermistor max temp error on boot

sevnspeed, 2020-12-15:
### Bug Description

  Buildig Marlin 2.05.2 or below thermistor number 3 or 66 from Dyze Design report at 876 degrees at room temp but in current versions max temp error at boot with max temp set to 9999 and thermal runaway off.  Cannot find a version of Marlin I can use successfully with my thermistors both of which are cartridge style thread in so don't know of a temporary substitute.  Board is Bigtreetech SKR 1.4 turbo and I have tried 3 new boards.  Previous board was Creality 10s pro V2 on marlin 2.0.5.2 tiny machines version and worked great with the 66 thermistor but layer shift issues which went away with Klipper but want to run Marlin.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[currentbugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5692791/currentbugfix.zip)
[2.0.5.2 skr.zip](https://github.com/MarlinFirmware/Marlin/files/5692796/2.0.5.2.skr.zip)
[crealityboard.zip](https://github.com/MarlinFirmware/Marlin/files/5692800/crealityboard.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Complie marlin with LPC1769
2. Complie Marlin with either thermistor 3 or 66
3. [and so on...]

**Expected behavior:**

Useable thermistors

**Actual behavior:**
  EIther 864 C at room temp reported or max temp error at boot.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-12-15:
You should not set max temp > highest temperature in the thermistor table.  as clearly this is value that is not reachable.
 table 66 highest temperature is 850
table 3 highest temperature is 864

ellensp, 2020-12-15:
I suspect you simply have a shorted thermistor 
Unplug the thermistor form the controller. Measure the resistance through the cables with a multi meter
type 66 Resistance at 20°C is 5.2 MΩ
type 3 Resistance at 25°C is 100KΩ

rflulling, 2020-12-15:
link to the thermal resistor you are working with, or spec sheet?

sevnspeed, 2020-12-16:
Thanks, I have tested with a multimeter and both are close to specs at room temp and the 500C was working great with Creality board and also with kipper, actually reported under 0C at room temp but once starting to heat it was fine, I've been trying to get the new board working for a month.  Spec sheets on www.dyzedesign.com but having trouble accessing the site atm.  The fact that 2.0.7 or 2.08 or whatever is current reports immediate max temp with max temp at 999C with both, the 300 being brand new and across 3 new boards while 2.0.5.3 is still wrong but reports 853 no max temp error seems odd to me. I tried switching to skr pro 1.2 but 2209 is not working with UART.  I just want 2209 UART with my thermistors to work, I do not care what board it is.  Klipper did work good but had to fine tune z offset after every power on and after a complete print wanted to home before moving which would run into part.....

sevnspeed, 2020-12-16:
Also the lcd screen isn't working on the skr pro and last firmware attempt made the sd card undetectable by my computer so afraid to try again.

Tonight I'll try editting klipper for the skr board and see how the temps report their and post the results.  I had to change the connectors for the new board but not the pins, only the locking clip was different, I don't have a depinning tool so the swap wasn't easy or I'd swap back to verify.

sevnspeed, 2020-12-19:
Proven to not be a hardware issue below:

Couldn't get klipper working but compiled Marlin 2.01 with the same settings with the 500C.  Had to change pins for the 1.4 to define lpc1769 since no turbo in this build, did not edit any stepper or movement settings but loaded and temp sensor reads 20C doing an autotune at 200 right now and screen is showing from 181 to 240.
[marlin2.01.zip](https://github.com/MarlinFirmware/Marlin/files/5720253/marlin2.01.zip)

closed 2021-01-27 by github-actions[bot]