MarlinFirmware/Marlin #20478: [BUG] #define RESTORE_LEVELING_AFTER_G28 causes failure to compile in Bugfix

bill-orange, 2020-12-15:
### Bug Description

The following combination of commented and uncommented lines causes a failure to compile.

```
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28
```
with the error

Marlin\src\gcode\calibrate\G28.cpp:438:62: error: 'leveling_was_active' was not declared in this scope; did you mean 'leveling_is_valid'?

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5696991/Marlin.zip)

### Steps to Reproduce

1.Compile with noted lines commented and uncommented
2.Note failure

**Expected behavior:**

Successful compile

**Actual behavior:**

Error as noted above or possibly sanity check error

#### Additional Information

```
//#define RESTORE_LEVELING_AFTER_G28
#define ENABLE_LEVELING_AFTER_G28
```

Gives the somewhat concerning warning:

Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:244:16: warning: unused variable 'leveling_restore_state' [-Wunused-variable]
  244 |     const bool leveling_restore_state = ENABLED(ENABLE_LEVELING_AFTER_G28) || TERN0(RESTORE_LEVELING_AFTER_G28, planner.leveling_active);
      |                ^~~~~~~~~~~~~~~~~~~~~~

Commenting out both lines also results in a failure

bill-orange, 2020-12-15:
@qwewer0 That does seem to address the problem, but I think I will wait for this to get sorted out before trying a proposed fix.

qwewer0, 2020-12-15:
I would wait too.

GMagician, 2020-12-15:
You have to enabled it and set it to true or false

bill-orange, 2020-12-15:
@GMagician I have no idea what you mean by "it".  Can you be more specific?

GMagician, 2020-12-15:
My fault, I wrongly read your post, sorry

bill-orange, 2020-12-16:
@sjasonsmith Actually it does cause a failure to compile in these circumstances:

```
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28
```

with the error

Marlin\src\gcode\calibrate\G28.cpp:438:62: error: 'leveling_was_active' was not declared in this scope; did you mean 'leveling_is_valid'?

This is described in my original post along with the circumstance that gives a scary warning.

Are you sure the title should be changed?

sjasonsmith, 2020-12-16:
Sorry about that. I encountered the warning several times but never a failure. When I checked your description it was one block of output but apparently not the other.

steljwagh, 2020-12-16:
The original startpost is faulty?

Because this works fine:
```
//#define RESTORE_LEVELING_AFTER_G28
#define ENABLE_LEVELING_AFTER_G28
```
But
```
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28
```
gives the error:
Marlin\src\gcode\calibrate\G28.cpp:445:62: error: 'leveling_restore_state' was not declared in this scope

GMagician, 2020-12-16:
#define ENABLE_LEVELING_AFTER_G28 should be defined with true or false. But the error is not correct anyway

bill-orange, 2020-12-16:
@GMagician 

```
//#define RESTORE_LEVELING_AFTER_G28 
#define ENABLE_LEVELING_AFTER_G28 true
```

results in the warning:

Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:244:16: warning: unused variable 'leveling_restore_state' [-Wunused-variable]
  244 |     const bool leveling_restore_state = ENABLED(ENABLE_LEVELING_AFTER_G28) || TERN0(RESTORE_LEVELING_AFTER_G28, planner.leveling_active);

GMagician, 2020-12-16:
Sorry, wrong copy &paste, RESTORE_LEVELING_AFTER_G28 must be true or false.
ENABLE_LEVELING_AFTER_G28 doesn't really use the variable, closed PR #20474 should fix that

bill-orange, 2020-12-16:
@GMagician 

```
#define RESTORE_LEVELING_AFTER_G28 true
//#define ENABLE_LEVELING_AFTER_G28 

```

results in:

Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:438:62: error: 'leveling_was_active' was not declared in this scope; did you mean 'leveling_is_valid'?

Hopefully #20474 wull fix that.

bill-orange, 2020-12-17:
@GMagician @sjasonsmith Okay I guess #20484 is closed.

```
#define RESTORE_LEVELING_AFTER_G28 true
//#define ENABLE_LEVELING_AFTER_G28
```
The selection above now compiles normally.   I suggest that the comment block directly above these lines state that **RESTORE_LEVELING_AFTER_G28 must be true or false.**  Its sure not obvious.

So far so good.  However:

```
//#define RESTORE_LEVELING_AFTER_G28 true
#define ENABLE_LEVELING_AFTER_G28
```

compiles but results in the warning:

```
Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:244:16: warning: unused variable 'leveling_restore_state' [-Wunused-variable]
  244 |     const bool leveling_restore_state = ENABLED(ENABLE_LEVELING_AFTER_G28) || TERN0(RESTORE_LEVELING_AFTER_G28, planner.leveling_active);

```

Is this harmless?  It probably needs fixing either way.  Sanity check errors for other defective selections seem to work fine.

sjasonsmith, 2020-12-17:
Changes are still ongoing in #20479.

I haven’t been paying attention to what is happening in that PR. You may want to go take a look and see whether you think it is making things better or not.

qwewer0, 2020-12-17:
I could reopen https://github.com/MarlinFirmware/Marlin/pull/20474, if it is a valid fix for the issue and if it is urgent, but I think https://github.com/MarlinFirmware/Marlin/pull/20479 should be the main goal.

bill-orange, 2020-12-17:
@qwewer0 sounds good to me.

GMagician, 2020-12-17:
#20479 should also fix all of these issues

bill-orange, 2020-12-17:
I see #20479 is closed now.  Hopefully, that resolves most issues.  Even with #20479 

```
//#define RESTORE_LEVELING_AFTER_G28 true
#define ENABLE_LEVELING_AFTER_G28
```

generates the warning 

```
Marlin\src\gcode\calibrate\G28.cpp: In static member function 'static void GcodeSuite::G28()':
Marlin\src\gcode\calibrate\G28.cpp:244:16: warning: unused variable 'leveling_restore_state' [-Wunused-variable]
  244 |     const bool leveling_restore_state = ENABLED(ENABLE_LEVELING_AFTER_G28) || TERN0(RESTORE_LEVELING_AFTER_G28, planner.leveling_active);
```

tpruvot, 2020-12-18:
made the PR #20499 for that

bill-orange, 2020-12-20:
Will this be rolled into BigFix shortly?  I have machine to which I am adding BL Touch.  I would like to wait for this PR before doing the  changes and recompile.

ETE-Design, 2020-12-22:
What is the difference for RESTORE_LEVELING_AFTER_G28 and ENABLE_LEVELING_AFTER_G28 and is it currect understod that ENABLE_LEVELING_AFTER_G28 dosen't make sence if you use UBL cause you have to define what slot to use if you have more that 1 mesh?

bill-orange, 2020-12-22:
Perfect compile after merging #20499.  I was going to close this issue but , come to think about it, I have the same question as @ETE-Design .

ETE-Design, 2020-12-24:
Mabye @thinkyhead can tell?

qwewer0, 2021-01-02:
@bill-orange @ETE-Design 
Before G28 the leveling state can be ON or OFF.
G28 will turn the bed leveling OFF for homing:
https://github.com/MarlinFirmware/Marlin/blob/b0585e13d84758e344f1b8bcadea32534f56f455/Marlin/src/gcode/calibrate/G28.cpp#L246
After G28, `RESTORE_LEVELING_AFTER_G28` will set the bed leveling state to what it was before G28,
and `ENABLE_LEVELING_AFTER_G28` will turn bed leveling ON regardless the previous state:
https://github.com/MarlinFirmware/Marlin/blob/5b33afb1dd07529945af1d9d85bd345bb9fb890c/Marlin/src/gcode/calibrate/G28.cpp#L443-L444
Hope that I answered the question.

bill-orange, 2021-01-02:
@qwewer0  @ETE-Design 
That answered my question.  

Thanks.  Closing this issue.

closed 2021-01-02 by bill-orange

ETE-Design, 2021-01-02:
@qwewer0 So when using ENABLE_LEVELING_AFTER_G28 you don't need to add M420 S1; to Start-GCode anymore? And when using UBL for leveling you only need G29 Lx to load the right slot?

sjasonsmith, 2021-01-02:
Also, what does ENABLE_LEVELING_AFTER_G28 do if there is no valid mesh to load?

qwewer0, 2021-01-02:
> @qwewer0 So when using ENABLE_LEVELING_AFTER_G28 you don't need to add M420 S1; to Start-GCode anymore? And when using UBL for leveling you only need G29 Lx to load the right slot?

That's how it should work.

> Also, what does ENABLE_LEVELING_AFTER_G28 do if there is no valid mesh to load?

Now that is a good question that I don't know that answer for.

steljwagh, 2021-01-03:
> 
> 
> Also, what does ENABLE_LEVELING_AFTER_G28 do if there is no valid mesh to load?

I think the question should be "what does ENABLE_LEVELING_AFTER_G28 do if there is no valid mesh to load?"

Because, ENABLE_LEVELING_AFTER_G28 should make a new grid/mesh while RESTORE_LEVELING_AFTER_G28 should take the saved grid/mesh.

qwewer0, 2021-01-03:
> Because, ENABLE_LEVELING_AFTER_G28 should make a new grid/mesh while RESTORE_LEVELING_AFTER_G28 should take the saved grid/mesh.

No, `ENABLE_LEVELING_AFTER_G28` and `RESTORE_LEVELING_AFTER_G28` does the same if leveling was on before G28, but if it was off then `ENABLE_LEVELING_AFTER_G28` will turn it on.
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/calibrate/G28.cpp#L443-L444
And I think if there is no or not valid mesh then leveling cannot be enabled, but that is only my idea...

bakaufman, 2021-01-03:
Some examples in the documentation of how these options would work are really needed. This is semantics at first glance but directly impacts your starting gcode.