MarlinFirmware/Marlin #20481: [FR] Confusing UBL Menu, especially probing manually

qurvento, 2020-12-15:
### Description

1. Confusing "**Mesh Edit**" **item** and "UBL Tools/**Edit Mesh**" **submenu**. "Mesh Edit" item is actually "Fine Tune Selected" and should be better renamed and moved to "UBL Tools/Edit Mesh" submenu to similar "Fine Tune" items.

2. Confusing "**Manually Build Mesh**" item **placement** for printers without bed probe.
"Manually Build Mesh" item is placed outside "Build Mesh" submenu, so the first impression is that "Build Mesh" submenu is for Auto Bed Levelling only, but there are other common tools as Fill-in, Continue and Invalidate. So "Manually Build Mesh" item should be better moved to "Build Mesh" submenu, and placed first. Those users who have probe will not notice, but those, who level manually, won't be confused by "Build Mesh (PLA)" and similar items, which don't work with "PROBE_MANUALLY" defined.

3. "**Fade Height**" is necessary parameter for activating UBL and shouldn't be the last in the list, it can be better placed near "Activate UBL" item so it won't be forgotten.

4. **"UBL Tools"** is the basic submenu and should be better placed right after "Step-By-Step UBL".

5. "UBL Tools/Edit Mesh/**Adjust Mesh Height**" **submenu** has only one item "Height Amount" and can be replaced by "Mesh Height Amount" item.

6. "**Output UBL Info**" submenu and "**Output Mesh Map**" item are placed in nonfunctional order and separately. They can be better placed nearby at the end of the list.

#### Additional Information

<pre>
// <b>SUGGESTED</b> UBL Menu

Unified Bed Leveling {
	Activate UBL
	<b>Fade Height</b>
	Step-By-Step UBL {}
	<b>UBL Tools</b> {
		Build Mesh {
			<b>Manually Build Mesh</b>
			Build Mesh (PLA)
			Build Mesh (ABS)
			Build Custom Mesh
			Build Cold Mesh
			Fill-in Mesh {
				Fill-in Amount
				Smart Fill-in
				Manual Fill-in
				Info Screen
			}
			Continue Bed Mesh
			Invalidate All
			Invalidate Closest
			Info Screen
		}
		Validate Mesh {
			Validate Mesh (PLA)
			Validate Mesh (ABS)
			Validate Custom Mesh
			Info Screen
		}
		Edit Mesh {
			Fine Tune All
			Fine Tune Closest
			<b>Fine Tune Selected</b>
			<b>Mesh Height Amount</b>
			Info Screen
		}
		Mesh Leveling {}
	}
	Mesh Storage {}
	<b>Output UBL Info</b>
	<b>Output Mesh Map</b> {
		Output for Host
		Output for CSV
		Off Printer Backup
	}
}
</pre>
<pre>
// <b>CURRENT</b> UBL Menu

Unified Bed Leveling {
	Activate UBL
	Step-By-Step UBL {}
	<b>Mesh Edit</b>
	Mesh Storage {}
	<b>Output Mesh Map</b>	{
		Output for Host
		Output for CSV
		Off Printer Backup
	}
	<b>UBL Tools</b> {
		Build Mesh {
			Build Mesh (PLA)
			Build Mesh (ABS)
			Build Custom Mesh
			Build Cold Mesh
			Fill-in Mesh {
				Fill-in Amount
				Smart Fill-in
				Manual Fill-in
				Info Screen
			}
			Continue Bed Mesh
			Invalidate All
			Invalidate Closest
			Info Screen
		}
		<b>Manually Build Mesh</b>
		Validate Mesh {
			Validate Mesh (PLA)
			Validate Mesh (ABS)
			Validate Custom Mesh
			Info Screen
		}
		<b>Edit Mesh</b> {
			Fine Tune All
			Fine Tune Closest
			<b>Adjust Mesh Height {
				Height Amount
				Info Screen
			}</b>
			Info Screen
		}
		Mesh Leveling {}
	}
	<b>Output UBL Info</b>
	<b>Fade Height</b>
}
</pre>

Vertabreak, 2021-04-08:
https://github.com/MarlinFirmware/Marlin/pull/21556
this might help i made a UBL Mesh Wizard that guides the user thru mesh creation with and without a probe.
a push button solution to make UBL easier to use.

Triangulix, 2022-07-30:
I can confirm that the whole UI regarding UBL is not self-explanatory. This alone wouldn't be a problem if there was documentation. Unfortunately there isn't.