MarlinFirmware/Marlin #20482: [BUG] (Add HOME_Z_FIRST option (#20113)  breaking bed leveling)

Mezabaru, 2020-12-16:
### Bug Description

Bed Leveling stops working on this commit for SKR Mini E3
https://github.com/MarlinFirmware/Marlin/commit/2282801172abdbce0b26b993d8d3be9ab4ae3649
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5699458/Marlin.zip)

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5699458/Marlin.zip)

### Steps to Reproduce

1. Flash Build using https://github.com/MarlinFirmware/Marlin/tree/f4e9f3654e9be66b5a2234642ccf736a8467dee3 and use these config https://github.com/MarlinFirmware/Configurations/tree/172189ca0f87fbd647cdd824ba9ceac879228152. prints like a charm.
2. https://github.com/MarlinFirmware/Marlin/tree/2282801172abdbce0b26b993d8d3be9ab4ae3649 flash build using this and bed leveling breaks.

**Expected behavior:**

Bed leveling should work.

**Actual behavior:**

Bi-linear bed leveling does not work.  My bed is fairly level, but the first layer does not adhere, despite not changing any slicer settings.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2020-12-16:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Mezabaru, 2020-12-16:
It didn't seem to go the first time, I'm going to verify my settings and calibrations and get back to you once I can fully confirm the state off the latest build.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5699752/Marlin.zip)

thisiskeithb, 2020-12-16:
The commits & PR you reference above are from November, but `bugfix-2.0.x` is updated daily, so that's why we ask you to test with it before reporting a bug.

Mezabaru, 2020-12-16:
looks like it was fixed within a few days! I'll go ahead and close this! I've done a few test prints and going fine at 150mm/s!

closed 2020-12-16 by Mezabaru