MarlinFirmware/Marlin #20483: [BUG] Ender 3 V2 UBL with BLTouch v3.1 cannot get a mesh to finish without failing.

Oninaig, 2020-12-16:
### Bug Description

Trying to execute a `G29 P1` will cause the probe to fail at some point during the mesh generation. It could be either near the beginning or right before it finishes. Sometimes it will fail and gracefully print the probe failure message to the log. Other times, the probe will trigger (turn red) but the z axis will keep pushing down, smashing into the bed until I am forced to kill the power to the printer.

In addition, the points in the mesh that it does generate do not make sense. The Ender 3 v2 comes with a glass bed which I have flipped over in order to use the flat backside. Out of the 4 manual leveling knobs on the bed, the top left is tightened _all the way_ and it still shows at least an offset of +.190 to +.300. There are also various points in the mesh that suggest that my flat glass bed is somehow filled with various craters?

Using an Ender 3 V2 with a BLTouch v3.1 with the latest marlin bugfix branch as of 12/14/2020.
### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5699607/Marlin.zip)

Enabled UBL, slowed down the probe speed to increase accuracy.

### Steps to Reproduce

1. Issue `G28`
2. Issue `G29 P1`

**Expected behavior:**

Mesh should generate all points it can without issue.

**Actual behavior:**

Probe with either fail at some random point during the generation, or it will trigger (red) but still crash into the bed anyway, forcing me to kill the power.

#### Additional Information

This is the second BLTouch I have purchased for this as I thought at first this was a wiring/bltouch issue. The bltouch and the wiring are brand-new and legitimate.

The BLTouch will also frequently fail the startup self-test, going into the red blinking failure mode. I have to restart the printer for this to go away and for it to actually finish the self-test.

The bltouch is wired the printer with the black/white wires going to the z-stop and the other 3 going to the appropriate bltouch port slots. My configuration is also configured to use the bltouch _as_ the z-stop. However, I have also tried just using the bltouch port by itself with all 5 wires with the same results.

Mezabaru, 2020-12-16:
That's normal behavior!

 I would recommend defining your bed positions, based on how far it can reach.
Double check your offsets as well you can do the following to ensure you can reach all of your bed to get a few extra probes in.

you can do this, to make sure your getting the full size of your bed reachable.

G28 X Y
// @section machine

// The size of the print bed
#define X_BED_SIZE 235 //max x length your nozzle can reach
#define Y_BED_SIZE 235 //max x length your nozzle can reach

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0 (you can put this into negatives or positives to make sure thats where your nozzle is).
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250

This is roughly how I was shown, so if another user can improve upon what I've said it may be of help!

kyraxyg, 2020-12-16:
I am also using 3D touch with UBL on Ender 3 v2. I don't think there is an issue described here as I can successfully produce mesh. I think you have a problem with the connection from board to BL Touch. So I advise you to recheck your cabling and use glue, something to make the cable stable.

FanDjango, 2020-12-16:
> The BLTouch will also frequently fail the startup self-test,

I know you said it is new, and it is a second new one even.

Please go into the LCD menus and run the self test from there, many times. Please report how well that works. Also please try the M48 command from a terminal (a few times or with M48 P50 for example).

Is there any chance of (even though it is new) that the movement of the plastic pin is somehow impaired? The above tests will show that up quicker than levelling.

Sometimes the little ferrite screw in the bltouch needs to be adjusted. This has been reported a few times, but the chances of that being a problem is low if you have TWO of them with the same problems. Unless you have mounted the BLTouch near a strong magnetic field.

> Probe with either fail at some random point during the generation, or it will trigger (red) but still crash into the bed anyway, forcing me to kill the power.

Try this one:
`#define ENDSTOP_NOISE_THRESHOLD 2`
I needed it on my Ender 5, similar symptoms. Random fails.

No more ideas at the moment

CRCinAU, 2020-12-17:
Also, its unusual if the Ender 3 V2 can use the full 235x235 - max reliable is about 225x225

jonny772, 2020-12-17:
I have recently had similar issues with a BL Touch 3.1 after updating the firmware from an earlier (foolishly didn't record this) to V2.0.X.14. 

Having tried different BL Touch units and wiring my new best guess at the cause is proximity of the BL data lines to current carrying wires. A re-route of the BL touch data wires has significantly helped but I cannot avoid the hot end power. I note that the behavior of the hot end during bed leveling has changed from disabling the heaters to not in the new release.

FanDjango, 2020-12-17:
I should have added 

```
/**
 * Enable one or more of the following if probing seems unreliable.
 * Heaters and/or fans can be disabled during probing to minimize electrical
 * noise. A delay can also be added to allow noise and vibration to settle.
 * These options are most useful for the BLTouch probe, but may also improve
 * readings with inductive probes and piezo sensors.
 */
#define PROBING_HEATERS_OFF         // Turn heaters off when probing
#if ENABLED(PROBING_HEATERS_OFF)
  //#define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy)
#endif
#define PROBING_FANS_OFF            // Turn fans off when probing
//#define PROBING_STEPPERS_OFF      // Turn steppers off (unless needed to hold position) when probing
#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors
```
to my above list of things to check.

These are some measures you can take to see if NOISE on the signal line is the cuplrit. This would cause spurious unexpected triggers, late or early triggers.

But the OP's description looked more like failing to DEPLOY the pin - which points more to the servo signal going to the probe. Of course, that could also be suffering from noise coming in, which would cause commands to the probe to be slightly mangled. This is less often seen though.

jonny772, 2020-12-17:
FanDjango, this is very helpful at least for my own issues, I will report back having made some changes

FanDjango, 2020-12-17:
Although only really intended for piezo probes, the
`#define DELAY_BEFORE_PROBING 200`
gives the probe some time to recongize a servo signal that was being mangled by heater pwm on the nozzle carriage cable. But your mileage may vary.

Some users will go so far as to use shielded cables for the probe wiring.

But I would check all my options first.

Oninaig, 2020-12-18:
Update: So I took the printer apart and rebuilt it, making sure to hot-glue the bltouch wires to the board and so far I am able to successfully produce a mesh. However, I am still having to adjust the bed-leveling knobs _after_ the mesh is generated because if I just generate the 10x10 mesh, the first layer test will show various parts of the bed being printed too close to the point where the nozzle is scratching the bed. This is also after manually leveling the center and all 4 corners with the paper test. 

I am thinking of switching to a bilinear mesh instead as I believe the ubl mesh is over-correcting at various points causing these issues. If I create the mesh and tweak it for my first print, I shouldn't have to keep creating new meshes every time to get the first layer to be even for each subsequent print but that is what I find myself having to do.

hamster65, 2020-12-18:
Do you have s curve acceleration enabled? If so, please try without it: https://github.com/MarlinFirmware/Marlin/issues/20105#issuecomment-748089680

FanDjango, 2020-12-18:
With no further data I cannot help. Although I would have hoped you could find some way to do a M48.

Why would this be interesting? Well, it removes the movement part of the problem, it probes in the same place many many times. It would help to isolate the problem, what is failing....

For example, if M48 works ALL of the time, but mesh probing not, one could go on from there.

Further steps would be to activate levelling debuging and looking at the output etc. etc. and to do further investigations.

But my crystal ball also has no LCD, I am touching it but getting no response.

FanDjango, 2020-12-18:
> Probe with either fail at some random point during the generation, or it will trigger (red) but still crash into the bed anyway, forcing me to kill the power.

> So I took the printer apart and rebuilt it, making sure to hot-glue the bltouch wires to the board and so far I am able to successfully produce a mesh

> I am thinking of switching to a bilinear mesh instead as I believe the ubl mesh is over-correcting at various points causing these issues

Thank you for keeping us up to date on this issue

Oninaig, 2020-12-19:
> Thank you for keeping us up to date on this issue

For the record, I was able to do an m48, here are my results:

```Recv: Finished!
Recv: Mean: 0.027750 Min: 0.021 Max: 0.032 Range: 0.011
Recv: Standard Deviation: 0.003444```

FanDjango, 2020-12-19:
How many M48's did you do? How many failed? Did they fail equally often as when you were probing a mesh?

Mezabaru, 2020-12-19:
Just want to throw in bilinear is giving me perfect results being in the start gcode as per @FanDjango<mailto:notifications@github.com> advice via a different channel.

[EXTERNAL]

sjasonsmith, 2020-12-20:
For the missed probes where it crashes into the bed, I strongly suspect a wiring issue. BLTouch wiring is quite fragile. It is easy to break the wires at the connects. Even newly installed connectors can be bad because the wires easily break if over-crimped. To test for wiring issues I suggest you _try_ to break your wires with a gentle tug. If they fall apart, that was your problem.

It sounds like maybe you already resolved that issue?

CRCinAU, 2020-12-20:
^^ This - and if you're just smooshing the dupont ends onto an XH2.54 socket, it is a crappy connection at best.

If you can get someone to crimp a 5 pin XH2.54 plug onto your BLTouch, then it'll likely never fail again...

sjasonsmith, 2020-12-22:
@Oninaig do you have any update on this, has it been resolved?

sjasonsmith, 2020-12-31:
Closing. Original submitter is unresponsive, and other users are not experiencing this issue on the same hardware.

closed 2020-12-31 by sjasonsmith

olartic, 2021-02-13:
I believe that I am having this problem, also. I can't do an M48 today (unknown command), on the most recent nightly, but probably can on some earlier version.

Yesterday's build gave me a blinking light and did nothing, today's build went through building a mesh, but threw some error after it went through UBL process, and came back to center and didn't give any data about the mesh.

Probe will work on some, but not all builds, but I have yet to get a build that I compile that works as expected. UBL used to not work at all, but now it gets through most of the process.

[2021.02.13.configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5976551/2021.02.13.configurations.zip)
[2021.02.13.Marlin 2.0.X bugfix BLTouch debug.txt](https://github.com/MarlinFirmware/Marlin/files/5976553/2021.02.13.Marlin.2.0.X.bugfix.BLTouch.debug.txt)