MarlinFirmware/Marlin #20484: [FR] (Check to ensure free save to save mesh data)

Mezabaru, 2020-12-16:
### Description
Basically say you want a 15x15 grid yeah? My suggestion is to add a check to cross reference average mesh data, with free space in EPROM, and through a warning if there will not be enough space to save the grid size necessary.  Just providing a check so the user will have enough space to store it.  Though I don't feel this should be enforced and should be a feature that can be disabled.

### Feature Workflow

1. Checks bed leveling method, if its linear, bi-linear, UBL proceed.
2. Checks the set grid size defined, if any other features are on that may additional increase mesh data.
3. compares against average spaced needed to save at-least 1 mesh.
4. Outputs warning telling user to reduce grid size or disable unneeded features 

#### Additional Information

Idea based on https://github.com/MarlinFirmware/Marlin/issues/20449

sjasonsmith, 2020-12-16:
UBL already calculates the number of meshes it can store, and should refuse to store one without space. As I mentioned after you posted to #20449, that is most likely a watchdog reboot due to slow access to that board's EEPROM.

closed 2020-12-16 by sjasonsmith

sjasonsmith, 2020-12-16:
I should have things set up to look at that issue. I will keep this suggestion in mind if I am wrong about my own watchdog assumption, and will re-open it if needed.