MarlinFirmware/Marlin #20486: [BUG] Clicking on Menu let the Fan stop, and runout detected, 13 UBL Points EEPROM safing failes (reboot while safing), Cure makes a Reboot when opening

Bastlwastl84, 2020-12-16:
Marlin Bugfix from before 1 hour

BTT SKR Mini E3 V2
TFT 35 E3 V3
ESP32-Cam
Neopixel
BTT Filament Runout Sensor (motion Sensor)
Happened in Marlin-Mode

Clicking on adjust Menu let the Part Cooling Fan pause at long Moves and sometimes parallel the Filament-Runout Sensor detect at same Time a runout, when opening the adjust Menu

.

since this Version (old was 14 Days old) Problems with false triggered Runout.

UBL with 13 Points let the Board reboot when saving, 12 Points working

Opening Cure let the Board reboot till closing Cura 4.8
[208 board.zip](https://github.com/MarlinFirmware/Marlin/files/5704205/208.board.zip)

kyraxyg, 2020-12-17:
I think you should create single issue for every other problem. The problem about 13*13 ubl seems to be like my issue  #20449.

Bastlwastl84, 2020-12-17:
Now partcooling fan workes only binary (fullspeed or off)

sjasonsmith, 2020-12-17:
We cannot act upon this report in it’s current form. Please report one bug at a time, and be sure to follow the complete template for each.

Some of these may be already reported as mentioned for the UBL issue.

If you have many issues, I suggest focusing on the most important issues first, rather than reporting them all at the same time.

closed 2020-12-17 by sjasonsmith